kyleisah/Klipper-Adaptive-Meshing-Purging #84: Problem with Adaptive Mesh and Klicky Probe

yviccc, 2023-05-06:
**Describe the issue**
When I run BED_MESH_CALIBRATE it simply attach then detaches the probe in sequence, I don't see any line in the code doind this
I followed every step correctly

my CFG files
[klackender.txt](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/11413364/klackender.txt)
[Adaptive_Mesh.txt](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/11413365/Adaptive_Mesh.txt)
[printer.txt](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/11413366/printer.txt)

cryjackel, 2023-05-06:
I´ve the same problem, the BED_MESH_CLAIBRATE sent the GCODE Attach_Probe and dock_probe without doing something and then i got the "Probe triggered prior to movement" error.
If i run ATTACH_PROBE oder DOCK_PROBE its runs w/o problems.

cryjackel, 2023-05-06:
If i attach Klicky by hand BED_MESH_CALIBRATE run with no errors but it will not dock the probe

kyleisah, 2023-07-06:
Try updating to the newest version

closed 2023-07-06 by kyleisah

BelgarionNL, 2023-07-06:
@kyleisah you closed this but turns out I also have the same problem with the latest version. it will not grab or park the probe. I have to manually add this in my startup script: attach_probe, Detach_probe regardless if I  change your boolean in the config True for klicky probes.

kyleisah, 2023-07-07:
You likely need to move some of your [include] calls to the TOP of printer.cfg, because KAMP is being included before the probe macros are being included or something like that.