kyleisah/Klipper-Adaptive-Meshing-Purging #91: Purge Line

choice3d, 2023-07-06:
I've uncommented the the purge line from the KAMP settings file and added it to the printer.cfg. But I don't get  a purge line or anything. No error messages either. Do I need I need to add something to the slicer or somewhere else? I've removed my purge line gcode from my slicer, was I not meant to do that?

I installed it all today using the newer method via ssh.

KAMP mesh works amazing btw, love it!

Cheers in advance

kyleisah, 2023-07-06:
Are you calling the Line_Purge macro in PRINT_START?

kyleisah, 2023-07-06:
There's also no klipper log file or anything. Those requests are in the issue template for a reason

choice3d, 2023-07-06:
No I'm not, I guess that's it. Should I add that to the end of my start gcode

choice3d, 2023-07-06:
Fixed and working well.

Cheers

closed 2023-07-06 by kyleisah