kyleisah/Klipper-Adaptive-Meshing-Purging #92: Error evaluating 'gcode_macro BED_MESH_CALIBRATE:gcode': UndefinedError: 'dict object' has no attribute 'polygon'

Johnn201, 2023-07-06:
**Describe the issue**
When going to print a file, or do a bed mesh calibrate I get this error: Error evaluating 'gcode_macro BED_MESH_CALIBRATE:gcode': UndefinedError: 'dict object' has no attribute 'polygon'

**What slicer are you using?** Superslicer
Prusaslicer, Superslicer, Cura, Orcaslicer...

**Troubleshooting**
What troubleshooting steps have you tried?

**Screenshots**
[image]

If applicable, add screenshots to help explain your problem.

**Klipper log file:**
Please add klipper log fil
[klippy (1).log](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/11966557/klippy.1.log)
e here...

**Sliced Gcode**
Please add sliced gcode file you are using here...

**Additional context**
Add any other context about the problem here.

kyleisah, 2023-07-06:
It looks like your gcode preprocessor isn't taking labeled object information and making it exclude object information.

Johnn201, 2023-07-06:
I don't understand it was working fine 1 day ago.

kyleisah, 2023-07-07:
```gcode

; object:{"name":"Dust_Shoe_Mandrill_TOP","id":"Dust_Shoe_Mandrill_TOP.stl id:0 copy 0","object_center":[112.500000,112.500000,0.000000],"boundingbox_center":[112.500000,112.500000,17.500000],"boundingbox_size":[115.980335,68.827911,35.000000]}
; plater:{"center":[112.500000,112.500000,0.000000],"boundingbox_center":[112.500000,112.500000,17.500000],"boundingbox_size":[115.980335,68.827911,35.000000]}

;TYPE:Custom
G28 ; home all axes
G1 Z5 F5000 ; lift nozzle
M107
G21 ; set units to millimeters
G90 ; use absolute coordinates
; Filament gcode
M117
            M190 S0
            M109 S0
            PRINT_START EXTRUDER_TEMP=210 BED_TEMP=60  MATERIAL=PLA+ 
                        
;_TOOLCHANGE 0
```

Look, as I mentioned before, your preprocessor is not defining objects when your gcode file is uploaded. This is not a KAMP issue.

closed 2023-07-07 by kyleisah

Johnn201, 2023-07-07:
Consider helpfull advice next time like maybe this, https://github.com/kageurufu/preprocess_cancellation?search=1