kyleisah/Klipper-Adaptive-Meshing-Purging #93: Movement outside of printable area

ksummers92, 2023-07-06:
On my V0.2, if the build plate is particularly full (but still enough room for purge), then the x_origin is set to 0, but the printer moves to a position less than 0 resulting in a print failure.

Slicer: SuperSlicer 2.5.59.2

I have tried changing the macro to set the x_origin/y_origin to 0.000 if x_origin is less than 0 in case it is a rounding error problem, but the same behavior results.

Applicable screenshot:
[image]

Klippy log:
[klippy (10).log](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/11968241/klippy.10.log)

Sliced GCode:
[side_strap_x4.zip](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/11968250/side_strap_x4.zip)

TomMeyer90, 2023-07-06:
Since the last update, the Voron purge is no longer available. Will be added again later.
you must select the line purge

ksummers92, 2023-07-06:
Ah I see. Just released yesterday. Guess I hadn’t seen the update from moonraker updater yet. Thanks!

closed 2023-07-06 by ksummers92