kyleisah/Klipper-Adaptive-Meshing-Purging #95: Unknown command:"PRINT_START" and Unknown command:"MESH_CONFIG"

Elchschmied, 2023-07-06:
**Describe the issue**
Startgcode no longer work. Leveling isn't done.

**What slicer are you using?**
Prusaslicer

**Troubleshooting**
I installed old macros and that works.
The new includes are made.
[image]

**Screenshots**
[image]

**Klipper log file:**
Please add klipper log file here...

**Sliced Gcode**
Please add sliced gcode file you are using here...

**Additional context**
Add any other context about the problem here.
[klippy(7).log](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/11970482/klippy.7.log)

kyleisah, 2023-07-07:
MESH_CONFIG is not a macro in KAMP.

closed 2023-07-07 by kyleisah

kkberg, 2023-07-07:
Sorry, this were the macros from last Version but I don‘t see a manual for the new commands. How do I have to procede with the new version? This Information is still missing.

Elchschmied, 2023-07-07:
okay, got it how to use. To easy for me, sorry.