kyleisah/Klipper-Adaptive-Meshing-Purging #96: Not grabbing probe when meshing.

BelgarionNL, 2023-07-06:
# The following variables are for those with a dockable probe like Klicky, Euclid, etc.                 # ----------------  Attach Macro | Detach Macro
variable_probe_dock_enable: True           # Set to True to enable the usage of a dockable probe.      # ---------------------------------------------
variable_attach_macro: 'Attach_Probe'       # The macro that is used to attach the probe.               # Klicky Probe:   'Attach_Probe' | 'Dock_Probe'
variable_detach_macro: 'Dock_Probe'         # The macro that is used to store the probe. 

but its still not picking up the probe when it starts meshing.

what is wrong?

BelgarionNL, 2023-07-06:
21:36
Probe triggered prior to movement
21:36
Probe triggered prior to movement
21:36
bed_mesh: relative_reference_index 4 is (109.99, 16.33)
21:36
15 | (10.0, 75.0) | (10.0, 75.0)
21:36
14 | (43.3, 75.0) | (43.3, 75.0)
21:36
13 | (76.7, 75.0) | (76.7, 75.0)
21:36
12 | (110.0, 75.0) | (110.0, 75.0)
21:36
11 | (110.0, 45.7) | (110.0, 45.7)
21:36
10 | (76.7, 45.7) | (76.7, 45.7)
21:36
9 | (43.3, 45.7) | (43.3, 45.7)
21:36
8 | (10.0, 45.7) | (10.0, 45.7)
21:36
7 | (10.0, 16.3) | (10.0, 16.3)
21:36
6 | (43.3, 16.3) | (43.3, 16.3)
21:36
5 | (76.7, 16.3) | (76.7, 16.3)
21:36
4 | (110.0, 16.3) | (110.0, 16.3)
21:36
3 | (110.0, -13.0) | (110.0, -13.0)
21:36
2 | (76.7, -13.0) | (76.7, -13.0)
21:36
1 | (43.3, -13.0) | (43.3, -13.0)
21:36
0 | (10.0, -13.0) | (10.0, -13.0)
21:36
bed_mesh: generated points
Index | Tool Adjusted | Probe
21:36
Generating new points...
21:36
KAMP adjustments successful. Happy KAMPing!
21:36
Adapted mesh bounds: (10.0, -13.0), (110.0, 75.0).
21:36
Fuzz amount is 0, mesh points not fuzzed.
21:36
Mesh margin is 0, margin not increased.
21:36
Default mesh bounds: (10.0, -13.0), (110.0, 75.0).
21:36
Adapted probe count: 4,4.
21:36
Default probe count: 4,4.
21:36
Algorithm: lagrange.
21:36
bed_mesh_calibrate

churls5495, 2023-07-06:
I'm having similar issues. KAMP was working prior to the v1.0 release, but with the new release it's not grabbing the probe.

gdachs, 2023-07-06:
I have made a pull request for a fix: https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/pull/97

kyleisah, 2023-07-07:
There is an issue template that needs to be followed when creating issues. There's a reason I ask for the klipper log file, it contains your entire config. Please attach a klipper log file.

BelgarionNL, 2023-07-07:
@kyleisah I understand, unfortunately I am overhauling my entire config structure so cant give it at the moment. perhaps @gdachs @churls5495 can provide you one.

in the meantime I have used @gdachs solution and it worked.

BelgarionNL, 2023-07-07:
[klippy (10).log](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/11983536/klippy.10.log)

churls5495, 2023-07-07:
gdachs fix also worked for me.

kyleisah, 2023-07-07:
@BelgarionNL Does running BED_MESH_CALIBRATE on it's own in the console pick up the probe and put it back?

B-Man1, 2023-07-08:
I believe the following lines are missing from KAMP/Adaptive_Meshing.cfg

{% set probe_dock_enable = printer["gcode_macro _KAMP_Settings"].probe_dock_enable | abs %}
    {% set attach_macro = printer["gcode_macro _KAMP_Settings"].attach_macro | string %}
    {% set detach_macro = printer["gcode_macro _KAMP_Settings"].detach_macro | string %}

For now a work around is to run "attach_probe" manually prior to running a besh mesh calibration and "detach_probe" manually afterwards

kyleisah, 2023-07-08:
> I believe the following lines are missing from KAMP/Adaptive_Meshing.cfg
> 
> {% set probe_dock_enable = printer["gcode_macro _KAMP_Settings"].probe_dock_enable | abs %} {% set attach_macro = printer["gcode_macro _KAMP_Settings"].attach_macro | string %} {% set detach_macro = printer["gcode_macro _KAMP_Settings"].detach_macro | string %}
> 
> For now a work around that worked for me is to run "attach_probe" manually prior to running a besh mesh calibration and "detach_probe" manually afterwards

Fuck. Me. You're right. How on earth did I miss that???

kyleisah, 2023-07-08:
Alright, they've been fixed and added to the newest release, updating should fix this.

B-Man1, 2023-07-08:
I also noticed that the status LED section that was present in the previous version of adaptive meshing is no longer present in this version.  I do have to say that this version certainly seems to fulfill its brief of being simpler and cleaner and easier to implement and update etc.  Adaptive meshing is truly awesome and I'm really appreciative of its creation.  Thank you so much!

kyleisah, 2023-07-08:
> I also noticed that the status LED section that was present in the previous version of adaptive meshing is no longer present in this version. I do have to say that this version certainly seems to [...]

Yeah, that was intentional. I saw a lot of users not using that, so I opted to remove it because the macro can just be called to change the LEDs before BED_MESH_CALIBRATE anyway. Figured it was best not to over-complicate the config and just keep it related to meshing and purging. 😅

tranhuunhan82, 2023-07-08:
After the upgrade I have the problem that after doing adaptive mesh the z coordinate is wrong when doing clean nozzle :|

kyleisah, 2023-07-08:
@tranhuunhan82 got a klipper log?

tranhuunhan82, 2023-07-08:
[klippy.log](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/11990504/klippy.log)

kyleisah, 2023-07-08:
@tranhuunhan82 is this happening when you are printing, or only when you were doing a bed mesh, and then running the clean_nozzle command?

In your print_start, BED_MESH_CALIBRATE happens *after* cleaning the nozzle. But if you were doing CLEAN_NOZZLE after doing BED_MESH_CALIBRATE, without doing z_calibration beforehand, then your offsets are going to be pretty crazy.

You also have a lot of leftover configs from the old KAMP, I recommend removing them.

tranhuunhan82, 2023-07-08:
I tried just using bed_mesh_calibrate and then executing command G1 z2 then z=-4.xx. Nothing printed yet

kyleisah, 2023-07-08:
so bed_mesh_calibrate worked fine, and what you did after caused the issue you're seeing. I suggest you comb through your config and remove any garbage. KAMP doesn't change any offsets, it makes a mesh. I'm closing this issue.

closed 2023-07-08 by kyleisah

tranhuunhan82, 2023-07-08:
OK, i'm understood

BelgarionNL, 2023-07-08:
bed_mesh_calibrate does pick up a probe etc. just FYI but since you already closed it I will try it again :)

thanks!

B-Man1, 2023-07-13:
This is probably not the place to put a feature request like this, but is it possible to size the adaptive bed mesh based on the dimensions of the first layer rather than the dimensions of the entire object?  I often print parts that have small first layers and large upper layers, but the bed mesh only needs to be determined based on the first layer so it would be more accurate and faster if the adaptive mesh was sized off the first layer rather than the entire object

kyleisah, 2023-07-14:
> This is probably not the place to put a feature request like this, but is it possible to size the adaptive bed mesh based on the dimensions of the first layer rather than the dimensions of the [...]

This would only *maybe* save you a couple seconds when probing. KAMP sizes the area based on objects for a reason, I wanted to avoid having to pass any params from the slicer to the macro. That being said, I can probably make you something, but I will not provide support for it.

B-Man1, 2023-07-14:
It only came up on the part below.  The time savings is not the benefit I see, though is there.  The main issue with this particular part is that the adaptive mesh covered the entire bed when only ~1/4" of it was occupied by the first layer, so the mesh was not as accurate as it could be in the area that it was needed.   In this case the 5x5 mesh (25 points) only had roughly 3x3 (9 points) in the area needed, I could up the mesh density, but that would take a really long time as a lot of other unneeded points would be added to the whole bed.

[image]
[image]

This is admittedly not a common print scenario, certainly something I can work with (or around if needed) and not worth making me a custom thing that would not be supported for, though I appreciate the offer.  If meshing based on first layer size rather than object size is easy to do then it might be beneficial for future, supported updates.  I am not a programmer so don't know the pros and cons of pashing parameters between the object and the slicer vs searching and puling values out of the gcode file directly etc etc, I just thought I'd share this unique use case with you to see what sparked in your creativity. 

Thanks for taking a look

Brian

kyleisah, 2023-07-14:
Well, it's not *just* the first layer we're worrying about. Meshes usually fade to zero after a set distance which is usually 10mm. So *50 layers* for typical print jobs.