kyleisah/Klipper-Adaptive-Meshing-Purging #100: Getting Internal error on command:"BED_MESH_CALIBRATE" after last version

patrezp, 2023-07-07:
**Describe the issue**
Getting this after the last update, it was working fine before:
2:58 PM
Internal error on command:"PRINT_START"
2:58 PM
Internal error on command:"BED_MESH_CALIBRATE"
2:58 PM
Internal error on command:"_BED_MESH_CALIBRATE"

**What slicer are you using?**
Superslicer

**Troubleshooting**
rollback to previous release

**Klipper log file:**
```
Algorithm: lagrange.
Default probe count: 7,7.
Adapted probe count: 3,3.
Default mesh bounds: (40.0, 40.0), (310.0, 310.0).
Mesh margin is 0, margin not increased.
Fuzz amount is 0, mesh points not fuzzed.
Adapted mesh bounds: (157.7, 157.7), (192.3, 192.3).
KAMP adjustments successful. Happy KAMPing!
Generating new points...
bed_mesh: generated points
Index |  Tool Adjusted  |   Probe
  0   | (157.7, 157.7)  | (157.7, 157.7)
  1   | (175.0, 157.7)  | (175.0, 157.7)
  2   | (192.3, 157.7)  | (192.3, 157.7)
  3   | (192.3, 175.0)  | (192.3, 175.0)
  4   | (175.0, 175.0)  | (175.0, 175.0)
  5   | (157.7, 175.0)  | (157.7, 175.0)
  6   | (157.7, 192.3)  | (157.7, 192.3)
  7   | (175.0, 192.3)  | (175.0, 192.3)
  8   | (192.3, 192.3)  | (192.3, 192.3)
Internal error on command:"_BED_MESH_CALIBRATE"
Traceback (most recent call last):
  File "/home/biqu/klipper/klippy/gcode.py", line 198, in _process_commands
    handler(gcmd)
  File "/home/biqu/klipper/klippy/gcode.py", line 135, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/biqu/klipper/klippy/gcode.py", line 135, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/biqu/klipper/klippy/extras/bed_mesh.py", line 606, in cmd_BED_MESH_CALIBRATE
    self.update_config(gcmd)
  File "/home/biqu/klipper/klippy/extras/bed_mesh.py", line 575, in update_config
    self.print_generated_points(gcmd.respond_info)
  File "/home/biqu/klipper/klippy/extras/bed_mesh.py", line 405, in print_generated_points
    % (rri, self.points[rri][0], self.points[rri][1]))
IndexError: list index out of range
Transition to shutdown state: Internal error on command:"_BED_MESH_CALIBRATE"
Dumping gcode input 0 blocks
Dumping 20 requests for client 281473291150144
Virtual sdcard (42414): 'p/8fcO2ofoNEAQgAAAAASU\n; VORK5CYII=\n; thumbnail end\n;\n; \n\n; external perimeters extrusion width = 0.40mm\n; perimeters extrusion width = 0.40mm\n; infill extrusion width = 0.40mm\n; solid infill extrusion width = 0.40mm\n; top infill extrusion width = 0.40mm\n; first layer extrusion width = 0.40mm\n\n; object:{"name":"Shape-Box","id":"Shape-Box id:0 copy 0","object_center":[175.000000,175.000000,0.000000],"boundingbox_center":[175.000000,175.000000,7.500000],"boundingbox_size":[35.000000,35.000000,15.000000]}\n; plater:{"center":[175.000000,175.000000,0.000000],"boundingbox_center":[175.000000,175.000000,7.500000],"boundingbox_size":[35.000000,35.000000,15.000000]}\n\n;TYPE:Custom\n; M190 S0\n; M109 S0 ; uncomment to remove set&wait temp gcode added automatically after this start gcode\nM117\n\n\n; Pre-Processed for Cancel-Object support by preprocess_cancellation v0.2.0\n; 1 known objects\nEXCLUDE_OBJECT_DEFINE NAME=Shape_Box_id_0_copy_0 CENTER=174.989,177.006 POLYGON=[[157.7,157.7],[157.7,192.3],[192.3,192.3],[192.3,157.7]]\n'
Upcoming (43438): 'PRINT_START EXTRUDER=225 BED=60 CHAMBER=0\nM107\nG21 ; set units to millimeters\nG90 ; use absolute coordinates\nM83 ; use relative '
gcode state: absolute_coord=True absolute_extrude=True base_position=[0.0, 0.0, 0.0, 0.0] last_position=[175.0, 175.0, 30.0, 0.0] homing_position=[0.0, 0.0, 0.0, 0.0] speed_factor=0.016666666666666666 extrude_factor=1.0 speed=150.0
Reactor garbage collection: (664057.129296729, 663981.048901026, 0.0)
Internal error on command:"_BED_MESH_CALIBRATE"
Internal error on command:"BED_MESH_CALIBRATE"
Traceback (most recent call last):
  File "/home/biqu/klipper/klippy/gcode.py", line 198, in _process_commands
    handler(gcmd)
  File "/home/biqu/klipper/klippy/gcode.py", line 135, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/biqu/klipper/klippy/extras/gcode_macro.py", line 189, in cmd
    self.template.run_gcode_from_command(kwparams)
  File "/home/biqu/klipper/klippy/extras/gcode_macro.py", line 68, in run_gcode_from_command
    self.gcode.run_script_from_command(self.render(context))
  File "/home/biqu/klipper/klippy/gcode.py", line 213, in run_script_from_command
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/biqu/klipper/klippy/gcode.py", line 198, in _process_commands
    handler(gcmd)
  File "/home/biqu/klipper/klippy/gcode.py", line 135, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/biqu/klipper/klippy/gcode.py", line 135, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/biqu/klipper/klippy/extras/bed_mesh.py", line 606, in cmd_BED_MESH_CALIBRATE
    self.update_config(gcmd)
  File "/home/biqu/klipper/klippy/extras/bed_mesh.py", line 575, in update_config
    self.print_generated_points(gcmd.respond_info)
  File "/home/biqu/klipper/klippy/extras/bed_mesh.py", line 405, in print_generated_points
    % (rri, self.points[rri][0], self.points[rri][1]))
IndexError: list index out of range
Internal error on command:"BED_MESH_CALIBRATE"
Internal error on command:"PRINT_START"
Traceback (most recent call last):
  File "/home/biqu/klipper/klippy/gcode.py", line 198, in _process_commands
    handler(gcmd)
  File "/home/biqu/klipper/klippy/gcode.py", line 135, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/biqu/klipper/klippy/extras/gcode_macro.py", line 189, in cmd
    self.template.run_gcode_from_command(kwparams)
  File "/home/biqu/klipper/klippy/extras/gcode_macro.py", line 68, in run_gcode_from_command
    self.gcode.run_script_from_command(self.render(context))
  File "/home/biqu/klipper/klippy/gcode.py", line 213, in run_script_from_command
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/biqu/klipper/klippy/gcode.py", line 198, in _process_commands
    handler(gcmd)
  File "/home/biqu/klipper/klippy/gcode.py", line 135, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/biqu/klipper/klippy/extras/gcode_macro.py", line 189, in cmd
    self.template.run_gcode_from_command(kwparams)
  File "/home/biqu/klipper/klippy/extras/gcode_macro.py", line 68, in run_gcode_from_command
    self.gcode.run_script_from_command(self.render(context))
  File "/home/biqu/klipper/klippy/gcode.py", line 213, in run_script_from_command
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/biqu/klipper/klippy/gcode.py", line 198, in _process_commands
    handler(gcmd)
  File "/home/biqu/klipper/klippy/gcode.py", line 135, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/biqu/klipper/klippy/gcode.py", line 135, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/biqu/klipper/klippy/extras/bed_mesh.py", line 606, in cmd_BED_MESH_CALIBRATE
    self.update_config(gcmd)
  File "/home/biqu/klipper/klippy/extras/bed_mesh.py", line 575, in update_config
    self.print_generated_points(gcmd.respond_info)
  File "/home/biqu/klipper/klippy/extras/bed_mesh.py", line 405, in print_generated_points
    % (rri, self.points[rri][0], self.points[rri][1]))
IndexError: list index out of range
MCU 'mcu' shutdown: Command request
```

**Sliced Gcode**
Please add sliced gcode file you are using here...
[cube-Voron_v2_350_afterburner 0.4 nozzle - test-Rosa - PLA Rainbow Silk.gcode.zip](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/11982420/cube-Voron_v2_350_afterburner.0.4.nozzle.-.test-Rosa.-.PLA.Rainbow.Silk.gcode.zip)

**Additional context**
Add any other context about the problem here.

kyleisah, 2023-07-07:
Have you updated klipper? There is something in the lot about RRI failing, and RRI has been deprecated and zero_reference_point is replacing it.

patrezp, 2023-07-07:
Not really, I was just reading about that. Will try to update klipper along with zero reference position and try again

patrezp, 2023-07-07:
I wanna keep my old klipper for now, but you're right @kyleisah it is related to zero_reference_position
Could you make sure that it is mentioned in readme that from this version of KAMP zero_reference_position needs to be configured in Klipper?

kyleisah, 2023-07-07:
> I wanna keep my old klipper for now, but you're right @kyleisah it is related to zero_reference_position Could you make sure that it is mentioned in readme that from this version of KAMP zero_reference_position needs to be configured in Klipper?

No, as zero_reference_position is only required for printers that have a z endstop **as well as** a probe. This will depend on the end-user knowing how to correctly configure their machine.

closed 2023-07-07 by kyleisah