kyleisah/Klipper-Adaptive-Meshing-Purging #102: Is KAMP_Settings.cfg supposed to be read only?

numindast, 2023-07-07:
**Describe the issue**
KAMP_Settings.cfg is read only in mainsail

**What slicer are you using?**
N/A, but SS and Cura

**Troubleshooting**
Edit KAMP_Settings.cfg using nano via SSH

**Screenshots**
[image]

**Klipper log file:**
Error evaluating 'gcode_macro BED_MESH_CALIBRATE:gcode': jinja2.exceptions.UndefinedError: 'extras.gcode_macro.GetStatusWrapper object' has no attribute 'gcode_macro _KAMP_Settings'

**Sliced Gcode**
n/a

**Additional context**
Also found I needed to move KAMP_Settings from the KAMP directory to ~/printer_data/config for the includes within KAMP_Settings to work as well.
Also had just updated Klipper/Mainsail/Moonraker (etc) from 11.0.200 as well. I do have a RRI that is generating a warning but haven't fixed that yet.

BelgarionNL, 2023-07-07:
when it install at least for me you get one in your config folder where klipper is. and the template which is readonly will stay in the KAMP folder.

maybe thats the reason that attach and dock probe isnt working because the settings arent read properly.

kyleisah, 2023-07-07:
That is intended yes. That way moonraker will say it's clean, and you'll have a copy as backup in case something gets fudged. The docking/undocking is another issue unrelated to KAMP.

closed 2023-07-07 by kyleisah