kyleisah/Klipper-Adaptive-Meshing-Purging #103: Quick jerk moves before purge line causing missed steps and printer to crash

dviera421, 2023-07-07:
**Describe the issue**
A clear and concise description of what the bug is...
When it goes to do the Purge Line it jerks around real quick losing steps and prints in the wrong area of the bed. Does not happen all the time, but most of the time. 

**What slicer are you using?**
Prusaslicer, Superslicer, Cura, Orcaslicer...
Prusaslicer

**Troubleshooting**
What troubleshooting steps have you tried?
Reviewed config files, adjusted Bed Mesh details in printer.cfg...played around with bed mesh and travel speeds. 

**Screenshots**
If applicable, add screenshots to help explain your problem.
No screenshots 

**Klipper log file:**
Please add klipper log file here...
Tried to upload Klippy log but got error "File size too big: 25 MB are allowed, 46 MB were attempted to upload." 

**Sliced Gcode**
Please add sliced gcode file you are using here...
"we don't support that file type"

**Additional context**
Add any other context about the problem here.

dviera421, 2023-07-07:
[image]

dviera421, 2023-07-07:
...other config info..

[bed_mesh]
speed: 250
horizontal_move_z: 5
mesh_min: 30,30
mesh_max: 220,220
probe_count: 5,5

[probe]
pin: PG15   
x_offset: 0
y_offset: 0
#z_offset: 6.42
speed: 10
lift_speed: 10
samples: 3
samples_result: median
sample_retract_dist: 1.5
samples_tolerance: 0.01
samples_tolerance_retries: 10

dviera421, 2023-07-07:
https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/assets/84048619/980bfe8b-740c-4a2e-9f68-9b76bd5d004c

kyleisah, 2023-07-07:
You likely have your max_velocity too high. That move uses your max velocity as a travel speed (that's what it's intended for). So try lowering that.

dviera421, 2023-07-07:
That was the trick, I had it set unusually high still :) thank you!

closed 2023-07-07 by dviera421