kyleisah/Klipper-Adaptive-Meshing-Purging #104: Adaptive bed mesh is not adapting to the print size

sdkrocks, 2023-07-07:
**Describe the issue**
Not matter what I do, it seems to probe the whole bed. Before starting it prints out this:

```
3:10 PM
bed_mesh: generated points
Index | Tool Adjusted | Probe
3:10 PM
Generating new points...
3:10 PM
KAMP adjustments successful. Happy KAMPing!
3:10 PM
Adapted mesh bounds: (35.0, 35.0), (290.0, 270.0).
3:10 PM
Fuzz amount is 0, mesh points not fuzzed.
3:10 PM
Mesh margin is 0, margin not increased.
3:10 PM
Default mesh bounds: (35.0, 35.0), (290.0, 270.0).
3:10 PM
Adapted probe count: 7,7.
3:10 PM
Default probe count: 7,7.
3:10 PM
Algorithm: bicubic.
```

**What slicer are you using?**
Prusaslicer, Superslicer, Cura, Orcaslicer...

**Troubleshooting**
I made sure to be on the most updated version of Cura (5.4) as I know the older one had an issue with object labels
I added M117 as the first gcode command in cura before START_PRINT
Made sure everything is imported and enabled

**Klipper log file:**
Please add klipper log file here...
[klippy.log](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/11985255/klippy.log)

**Sliced Gcode**
Please add sliced gcode file you are using here... 
(had to zip the gcode file for Github)
[gcode.zip](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/11985259/gcode.zip)
**Additional context**

Add any other context about the problem here.
In case it matters, I have multiple printers setup using Kiauth. I have made sure each have the correct config

kyleisah, 2023-07-07:
```gcode
;Generated with Cura_SteamEngine 5.4.0
M82 ;absolute extrusion mode
M117 Starting Print
START_PRINT T_BED=70 T_EXTRUDER=205.0 PRINT_MIN=%MINX%,%MINY% PRINT_MAX=%MAXX%,%MAXY%
G92 E0
G92 E0
G1 F1500 E-2
;LAYER_COUNT:119
;LAYER:0
M107
G1 F300 Z0.45
G0 F9000 X98.649 Y111.209 Z0.45
```

For whatever reason, objects are not being generated by the preprocessor when you upload gcode to your machine.

KAMP works based on exclude object information, so you will need to fix that.

If Klipper was detecting objects upon upload, you would see something like this in your gcode file:
[image]

sdkrocks, 2023-07-07:
> ```gcode
> ;Generated with Cura_SteamEngine 5.4.0
> M82 ;absolute extrusion mode
> M117 Starting Print
> START_PRINT T_BED=70 T_EXTRUDER=205.0 PRINT_MIN=%MINX%,%MINY% PRINT_MAX=%MAXX%,%MAXY%
> G92 E0
> G92 E0 [...]

Thanks for taking a look. Thinking about it, I guess that makes sense as I am using a network drive for the gcode files. So in this case, since I am not using the upload function and just printing from the network folder, I assume it would not get processed?

So in this case, if I wanted it to work and continue using the network drive, I would have to use something like the preprocessor for Cura. Or instead of printing right off the drive, I should just use the upload function instead. Did I get that correct?

Thanks!

kyleisah, 2023-07-07:
I believe that makes sense. How are you printing from a network drive? I wonder if the preprocessor can monitor that folder?

kyleisah, 2023-07-07:
The only way to get it to be readable for klipper would be to upload it to the klipper instance printing it, but there might be a workaround if I knew how you were printing from a network folder

kyleisah, 2023-07-07:
[image]

Something like this, but instead point that path to the folder on your network

sdkrocks, 2023-07-07:
> The only way to get it to be readable for klipper would be to upload it to the klipper instance printing it, but there might be a workaround if I knew how you were printing from a network folder

Currently I have the network drive mounted, and symlinked to each printer in the way that you said in your other reply. I did that so all my printers would have access to the same files. 

so in your example above, ~printer_1_data/gcodes is a symlink to the network mount.

AudreyAP, 2023-07-08:
The processing is only done by Moonraker when the file is uploaded to the Moonraker API. Nothing happens to files in the gcode folder. 

The solution for you is to use a postprocessing script to generate klipper-compatible object exclusion data: https://github.com/pedrolamas/klipper-preprocessor

AudreyAP, 2023-07-08:
Here is a simpler version compatible with most slicers: https://github.com/kageurufu/preprocess_cancellation

Both should work identically.

kyleisah, 2023-07-08:
@sdkrocks does @AudreyAP's solution work for you? I can close this up if you're happy with that solution

sdkrocks, 2023-07-08:
> @sdkrocks does @AudreyAP's solution work for you? I can close this up if you're happy with that solution

Yes this works! I'll try to get that working. Thanks for both yours and @AudreyAP 's help.

closed 2023-07-08 by kyleisah