kyleisah/Klipper-Adaptive-Meshing-Purging #99: Error evaluating 'gcode_macro BED_MESH_CALIBRATE:gcode'

locki-cz, 2023-07-07:
**Describe the issue**

Error evaluating 'gcode_macro BED_MESH_CALIBRATE:gcode': jinja2.exceptions.UndefinedError: 'extras.gcode_macro.GetStatusWrapper object' has no attribute 'gcode_macro _KAMP_Settings'

**What slicer are you using?**
Prusaslicer

**Troubleshooting**
What troubleshooting steps have you tried? Tryed fresh mainsal image. All steps for enable this KAMP done, it works fine on voron 2.4 but not on my ender with klipper.

**Klipper log file:**
Please add klipper log file here...

**Sliced Gcode**
Please add sliced gcode file you are using here...

**Additional context**
Add any other context about the problem here.

kyleisah, 2023-07-07:
KAMP_Settings is not found in your config. Did you forget to [include] it?

numindast, 2023-07-07:
I found I had to move KAMP_Settings.cfg to ~/printer_data/config for this error to go away. And updating printer.cfg to reflect that it's no longer in the KAMP folder. So my printer.cfg contains this now:

[image]

kyleisah, 2023-07-07:
> I found I had to move KAMP_Settings.cfg from the KAMP subfolder to ~/printer_data/config for this error to go away. And updating printer.cfg to reflect that it's no longer in the KAMP folder. So my printer.cfg contains this now:

That is a step in the instructions, to copy the file to your config directory.

Thank you for describing the other necessary steps.

closed 2023-07-07 by kyleisah