kyleisah/Klipper-Adaptive-Meshing-Purging #106: Not able to move purge line

patofoto, 2023-07-08:
**Describe the issue**
Just setup KAMP to use the KAMP_Settings.cfg file outside of the KAMP folder. I used to use this line in my START_PRINT to purge at a specified position:

`SETUP_LINE_PURGE X_DEFAULT=10 Y_DEFAULT=10`

**What slicer are you using?**
Orcaslicer

**Troubleshooting**
Using the above line in my START_PRINT g-code.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Klipper log file:**
Please add klipper log file here...

**Sliced Gcode**
Please add sliced gcode file you are using here...

**Additional context**
Would love to be able to place the purge line where I want it. Also it would be better for me if I was able to orient the line. It prints on the x axis (horizontal) and would love bo be able to choose to print on the y axis (vertically).

kyleisah, 2023-07-08:
If you want the purge line to be at a specific point, why are you using **adaptive** purging? 

I think you should just make a purge line macro and use that, rather than defeat the nature of an adaptive purge lol.

patofoto, 2023-07-08:
@kyleisah Thanks for your answer. You are right. I started using KAMP a while ago and needed to adjust things at some point. Got used to using KAMP and lost sight of why I was using it. Being able to change its orientation to purge on the Y axis would come in handy sometimes.

kyleisah, 2023-07-08:
Well, KAMP *can* do that, but only if there is no room in the front of the bed to purge. If that happens, it'll move to the side and purge vertically. 

Personally I don't see why anyone wouldn't just use an adaptive purge. Doing a purge in the same spot over and over causes wear on your build surface, and KAMP helps to extend the life of it. 

I'm going to close this issue as there's no apparent *problem* with KAMP, but I will keep in mind that maybe it would be benefitial to have some sort of param to pass to the purges to define a location.

Thank you!

closed 2023-07-08 by kyleisah

patofoto, 2023-07-08:
@kyleisah Thank you so much for you time.