kyleisah/Klipper-Adaptive-Meshing-Purging #107: Easier / cleaner way to reference the settings macro

AudreyAP, 2023-07-08:
Not sure if you intentionally chose to use the longer way, but you can replace this: 

```
{% set verbose_enable = printer["gcode_macro _KAMP_Settings"].verbose_enable | abs %}
```

With this: 

```
# Only needed once at the top of file
{% set kamp_settings= printer["gcode_macro _KAMP_Settings"] %}

{% set verbose_enable = kamp_settings.verbose_enable | abs %}

```

I've found that it makes working on my macros so much easier. Much less copy-pasting!

kyleisah, 2023-07-08:
Huh, I had no idea you could do it like that. I've always just done it the hard way I guess. 😅 I'll have to play around with that a bit!

godbnc, 2023-07-10:
Yes working :) tenks.. update with that:) original no work good i tested..

kyleisah, 2023-10-07:
@AudreyAP finally getting around to making these changes per your suggestion. I'm a huge fan of this method over the previous. When I finally get around to updating everything I'll close this up. Thanks again!

kyleisah, 2023-11-02:
All changes were made and merged with PR #216

closed 2023-11-02 by kyleisah