kyleisah/Klipper-Adaptive-Meshing-Purging #108: Extrusion mode change as a result of purge

ksummers92, 2023-07-08:
**Describe the issue**
Purge ends with a M82 command which is changing the extrusion mode from the default (for me at least). 

**What slicer are you using?**
Superslicer

**Troubleshooting**
I added a M83 command after the purge command in start print macro and it fixed the problem. 

I can add Klipper log and gcode file if needed, but this is a more general problem. 

I think checking for the extrusion mode at the start of the purge macros and then ending the macro with the applicable M82/83 would help reduce confusion.

kyleisah, 2023-07-08:
Ah right, this is something I meant to fix in dev. I was going to use gcode states so whatever it was before the purge would be what it was when the purge finishes. I'll work on this a bit and see what I can come up with.

kyleisah, 2023-07-14:
Just providing an update, I got around to adding gcode states to the purges to stop this from happening. 
[image]

The changes will be in the next update!

kyleisah, 2023-08-20:
This has been implemented in the current release. Please update!

closed 2023-08-20 by kyleisah