kyleisah/Klipper-Adaptive-Meshing-Purging #109: Bed Mesh Calibrate Crashes Klipper

claudermilk, 2023-07-09:
**Describe the issue**
Starting a print job with new KAMP using bed mesh crashes Klipper with this console output:
// Klipper state: Shutdown
09:28:41 
!! Internal error on command:"_BED_MESH_CALIBRATE"
09:28:41 
!! Internal error on command:"BED_MESH_CALIBRATE"
09:28:41 
!! Internal error on command:"PRINT_START"
09:28:41 
// Internal error on command:"_BED_MESH_CALIBRATE"
// Once the underlying issue is corrected, use the
// "FIRMWARE_RESTART" command to reset the firmware, reload the
// config, and restart the host software.
// Printer is shutdown
09:28:41 
!! Internal error on command:"_BED_MESH_CALIBRATE"
09:28:41 
// Internal error on command:"_BED_MESH_CALIBRATE"
// Once the underlying issue is corrected, use the
// "FIRMWARE_RESTART" command to reset the firmware, reload the
// config, and restart the host software.
// Printer is shutdown
09:28:41 
!! Internal error on command:"_BED_MESH_CALIBRATE"
09:28:42 
// Internal error on command:"_BED_MESH_CALIBRATE"
// Once the underlying issue is corrected, use the
// "FIRMWARE_RESTART" command to reset the firmware, reload the
// config, and restart the host software.
// Printer is shutdown
09:28:42 
!! Internal error on command:"_BED_MESH_CALIBRATE"

**What slicer are you using?**
Superslicer

**Troubleshooting**
Removed reference to KAMP_Settings.cfg, removed expanded mesh & fuzzy parameters. Removed bed_mesh_calibrate call in print_start. Only removing bed_mesh_calibrate call completely allows it to run.

Commented out moonraker update settings and reverted files to 0.1.2 to get back running.

**Screenshots**
n/a

**Klipper log file:**
[klippy.zip](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/11995724/klippy.zip)

**Sliced Gcode**
Please add sliced gcode file you are using here...

**Additional context**
I have been using the older version of KAMP successfully before now. All prerequisites have been set for a long time.

kyleisah, 2023-07-09:
Have you updated klipper? Seems to be getting an error when checking the reference index, which has been deprecated, and now zero_reference is the new "RRI". 

Also, it looks like you've got a ton of old leftover KAMP stuff in your config, might be good to clean all that old stuff out.

claudermilk, 2023-07-10:
I have not updated Klipper, Moonraker, or the OS because of the issues they have. So I'm on the last version before USB got borked.
I'll try going through and clearing all KAMP stuff and trying again. Maybe I missed something, but it should just be calling the settings macros, adaptive bed mesh, and adaptive purge in print_start.

kyleisah, 2023-07-10:
AFAIK the usb stuff got patched up, that was a bad udev package but I think that's all sorted now. 

Your log says RRI caused it, so I think updating klipper would fix it. If you installed with KIAUH you can easily roll back if that's not the case, but I got a pretty good feeling.

claudermilk, 2023-07-10:
OK, I updated everything to current version. Happily the only message I got was the RRI waring, so I commented that out. Removed and fully-reinstalled KAMP 1.1.1 from scratch. I did a bunch of long-overdue cleanup on the config files.

I am running a quick test print, and it happily ran the adaptive mesh, and parked where it should, now running the print after the Voron purge! Looks like it's all fixed up. I think the issue was that Klipper, Moonraker, and the OS need to be updated to the current version.

closed 2023-07-10 by claudermilk

kyleisah, 2023-07-10:
Fantastic to hear, I'm glad that's all working now!