kyleisah/Klipper-Adaptive-Meshing-Purging #111: Normal Bed Mesh dont appear in Fluidd

ManuelPeiro, 2023-07-11:
**Describe the issue**
A clear and concise description of what the bug is...

**What slicer are you using?**
Not slicer issue

**Troubleshooting**
I installed Kamp, and it works very well... But I have noticed that since I use it, a few days... If I make a manual Bed Mesh, it does not appear in Fluidd... I found it useful to check wl state of the bed

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Klipper log file:**
Please add klipper log file here...

**Sliced Gcode**
Please add sliced gcode file you are using here...

**Additional context**
Add any other context about the problem here.

kyleisah, 2023-07-11:
Can you please share a klipper log file?

kyleisah, 2023-07-12:
I'm going to close this issue. This is likely a client side issue and not a problem that KAMP is causing. All KAMP does is make a mesh. If fluidd or mainsail isn't showing the mesh, then the issue is likely with fluidd or mainsail.

closed 2023-07-12 by kyleisah

ManuelPeiro, 2023-07-12:
I have see later