kyleisah/Klipper-Adaptive-Meshing-Purging #112: New to KAMP and a little confused.

remizca, 2023-07-11:
So I've recently installed KAMP and I'm a little confused about the settings, when I go to the KAMP folder, everything is locked and I can't edit the settings for either Line purge or Voron purge. do I copy the content and make another file outside the kamp folder and change the include path on my printer.cfg? or do I just edit the settings in KAMP_Settings and it adjusts the values on the KAMP Folder files?
[image]
[image]

kyleisah, 2023-07-12:
Hey there! So, when you were doing the install, did you ssh into your pi and run the commands?

There was a command in there to copy the config file from KAMP to your printer.cfg directory. **That** file is the one that you use to adjust your settings. The one in the folder is meant to stay clean so when you make changes to it, moonraker doesn't mark the folder "Dirty". This is so when there's an update to KAMP, you'll know right away because you'll see it in the UI.

So in your first photo, just use the KAMP_Settings.cfg file you see there to make all your changes you want to KAMP without making the moonraker updater unhappy.

remizca, 2023-07-12:
Yes, I followed the instructions and tried a test print and it works.

Oh okay, thanks! I'll check KAMP_Settings further when i return home from work.

closed 2023-07-12 by kyleisah