kyleisah/Klipper-Adaptive-Meshing-Purging #113: New Line Purge Causes Line/Nozzle Collision When Moving to Print

Silky322, 2023-07-11:
**Describe the issue**
The new line purge macro sometimes results in the nozzle dragging back through the line when moving to the start of the print. Since the nozzle is hot it just cuts a divot in the line purge and dirties the nozzle instead of causing any significant issues. Previously you could set the line length along with the amount extruded to ensure the line was at or below the layer height, with the removal of line length in the config you always end up with a line of ~1mm height that the nozzle hits when it moves to the first layer height.

**What slicer are you using?**
Superslicer

**Troubleshooting**
Played with purge_height, purge_amount, and flow_rate. All seem to result in blobs of similar height but varying length instead of a longer line with a lower height. 

**Screenshots**
[image]

**Klipper log file:**
N/A

**Sliced Gcode**
Relevant section of gcode [here](https://pastebin.com/APm85fr9)

**Additional context**
For the print I was using for testing the voron purge seems to kinda fix the issue, mostly because it generates in a location where the move to the print start position doesn't intersect with the purge line, but there is still a small section on the edge. Overall this is minor as it just results in having to clean the nozzle a bit more frequently. 

I'm also working on seeing if there is a way to add some start gcode as well that would cause the nozzle to move towards the print start prior to moving to first layer height as that should also resolve the issue, but I'm not up to speed on superslicer variables so I'm not sure if that is supported.

Silky322, 2023-07-12:
Gotcha. After looking at the macro a bit more I kinda figured that would be the case. Guess I'll just clean the nozzle more often/use the voron purge instead. Thanks again for KAMP, it's a hell of a time saver.

quantumfrost, 2023-07-14:
> Yeah, this is something that I can't fix easily. It's caused by your slicer because the first line of gcode is setting your z height before moving to the print area. It also varies every print because you slicer dictates where the first line will start, so it's not possible to make this something you can define and adapt with, at least, not to my knowledge.
> 
> But, in all reality, it's not that big of a deal. The purge ends up in the trash anyway.

Couldn't you just move the nozzle (while maintaining the correct height) to the center of the print bounding box?

kyleisah, 2023-07-14:
> Couldn't you just move the nozzle (while maintaining the correct height) to the center of the print bounding box?

Yes, you could, but then you run the risk of oozing in the print area. Ideally I think the best way would be a rapid "wipe" move at the end of the purge line, just enough that it can run right back through it when moving to the first line from your slicer.

kyleisah, 2023-07-14:
@Silky322 Maybe I'm just being lazy. Let's improve this a bit. I have a revised Line_Purge macro I'd like you to try. We can work in a "string break" move and that should likely get the nozzle into an area far away enough to not hit the purge line.

Just comment out your current Line_Purge and add this one instead and give it a shot.

[Revised_Line_Purge.txt](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12046294/Revised_Line_Purge.txt)

kyleisah, 2023-07-14:
All I've done is add a 10mm rapid move at the end of the purge. This should break any stringing, as well as hopefully give you a good angle of attack to not hit the purge line.

Silky322, 2023-07-14:
That looks to have solved it. Tested with both the purge line on the x and y axis and both avoid crossing back over the line when it moves to start the print.

closed 2023-07-14 by Silky322

kyleisah, 2023-07-14:
@Silky322 sure thing, no worries. Yeah I had the move in a weird spot but after some thought I have it in a good spot. The updated macro is in my other branch for KAMP, so it will be added officially in the next update. Thanks for helping me out with this, and I'm glad it's helped you out as well!