kyleisah/Klipper-Adaptive-Meshing-Purging #114: Should the KAMP_Settings.cfg be tracked?

BassXX, 2023-07-12:
Hi there,

I noticed, when you change the KAMP Settings, the Update Manager in Klipper will tell you your version is compromised, as this leads to a dirty commit.

Is there a way to avoid this?

Thanks & Regards

kyleisah, 2023-07-12:
Hey there!

The KAMP_Settings.cfg file that lives inside of the KAMP folder is not meant to be edited, it is only there so it can be copied to the same directory your printer.cfg file is in. 

The copy that's in the same directory as printer.cfg is the one that can be edited and will keep the repo marked `clean`. 

So make sure you have a **copy** of KAMP_Settings.cfg in the correct directory and [include] that in your printer.cfg, and then you can use the copy of KAMP_Settings.cfg to do the other [includes] and setup that you need. 

It's in the installation instructions but is easy to miss. 😅

kyleisah, 2023-07-12:
After the ssh commands portion of the install documentation, there's a note. 

> Note: This will change to the home directory, clone the KAMP repo, create a symbolic link of the repo to your printer's config folder, and create a copy of KAMP_Settings.cfg in your config directory, ready to edit.

BassXX, 2023-07-12:
Ahhh! Got it, so I edited the wrong file!

Thanks for letting me know!

closed 2023-07-12 by BassXX