kyleisah/Klipper-Adaptive-Meshing-Purging #115: BED_MESH_CALIBRATE does not lift up on a first probing of a point

LogicNoo, 2023-07-12:
Hi,

I just updated the configs and noticed that while running BED_MESH_CALIBRATE macro for the fist probe of the point it does not lift Z. Probe point just drops and bounces on a bed then it is just carries on as normal until next point.  

At some point it fails an stops the print.

Version before the rework worked fine.

Also a shot vid to illustrate the issue.

https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/assets/92871559/73ba5d04-ab3d-4158-b941-e4d7791151f2

kyleisah, 2023-07-12:
Can you please upload a klipper log file?

LogicNoo, 2023-07-12:
[klippy (2).log](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12030991/klippy.2.log)

Sorry for got that logs exits.

kyleisah, 2023-07-12:
Ah dang, you don't have a print_start macro. Do you have a gcode file so I can see what your starting gcode is?

kyleisah, 2023-07-12:
I know it's not KAMP because all KAMP does is adjust your mesh points and nothing more. It doesn't do any move commands or anything.

What I *think* is happening is you either have a weird Z move in your start gcode, or, you have a slight error in your config.

At a cursory glance, here's what I see:

```yaml
[bed_mesh]
.
.
horizontal_move_z = 4

[bltouch]
.
.
sample_retract_dist = 5
```

I think if you make both of those values the same ( 5 I'm going to say) it should stop doing that.

kyleisah, 2023-07-14:
After you do `g28` and your axes home, can you do `m114` and give me the output? I'm curious what your Z is being set to after homing and if I can work out a fix for your setup.

LogicNoo, 2023-07-14:
m114 gives `X:165.500 Y:106.500 Z:10.000 E:0.000`

kyleisah, 2023-07-14:
> m114 gives `X:165.500 Y:106.500 Z:10.000 E:0.000`

Thanks for the quick response, let's try this:

```yaml
[bed_mesh]
.
.
horizontal_move_z = 10

[bltouch]
.
.
sample_retract_dist = 10
```

Does that improve it?

LogicNoo, 2023-07-14:
Sorry for a wait, had to go to work you see. That seems to make it work properly, but due to extra distance, probing is a bit slower. That can be fixed by playing around with values. Thanks you for a help and KAMP rocks!

closed 2023-07-14 by LogicNoo

huffraw, 2023-07-14:
I don't know if it improved it for him but it did for me.