kyleisah/Klipper-Adaptive-Meshing-Purging #117: (Question/Feature Request)

devsfan1830, 2023-07-12:
Working great but I have a functionality question that might be a feature request unless this is easily doable on my own. 

Is there a way to move a "cold" nozzle to the purge line starting position, or perhaps doing a nozzle scrub, before having the hotend heat to full temp and THEN doing the purge?

As it stands now, my nozzle starts at 150C, bed temp to whatever the material needs, to get a hot auto z adjustment. Then it moves to center, heats to full, then performs the purge line. Problem is, at least for PLA, it causes my nozzle to drip molten plastic before its able to purge. 

Before using KAMP, I would move the 150C nozzle to the bed where I was about to run my hard coded purge line, wait for full temp, then purge and print. I'd like to do the same with the adaptive purge line but not clear on how.

kyleisah, 2023-07-12:
Way ahead of you! Have you seen the Smart Park feature?

devsfan1830, 2023-07-12:
I did not! Whoops. I will try it out!

closed 2023-07-12 by devsfan1830