kyleisah/Klipper-Adaptive-Meshing-Purging #119: Voron Purge too close

maclakey, 2023-07-13:
Using Prusa Slicer.
Prusa skirt is set at 10mm from objects.
Voron Purge doesn't take the skirt into account when processing objects.
Voron Purge is set at 10mm from objects.
I try to set "variable_purge_margin: 20" in KAMP_settings.cfg but file is read only.
SSH in to edit KAMP_Settings.cfg, but now Update Manager says KAMP is "dirty".
I guess will leave it "dirty" until it's updated?

kyleisah, 2023-07-13:
You are not supposed to edit the settings file in the KAMP folder, as it will mark the repo dirty. 

In the instructions you are supposed to make a copy of it to the same directory as printer.cfg and include that one, and make your changes to the copy. 

Please give that a try and update me here.

maclakey, 2023-07-13:
Yep.  I'm a right fool.
After carefully reading through the instructions again, I saw what I did not do correctly.

kyleisah, 2023-07-13:
You're not the first, don't sweat it. 😅 did that fix it for you?

maclakey, 2023-07-13:
Definitely did.  Thank you!