kyleisah/Klipper-Adaptive-Meshing-Purging #120: "On/Off button" Temporary Disable adaptive bed mesh?

Ltek, 2023-07-13:
**Describe the issue**
Often I dont want to run a bed mesh with every print... Sometimes I'll do a full bed mesh and run multiple different prints - same day or across days. I only want to do an Adaptive Mesh when its been a while since my last 

**What slicer are you using?**
Orcaslicer and Cura

**Troubleshooting**
google searches

**Screenshots**
n/a

**Klipper log file:**
n/a

**Sliced Gcode**
n/a

**Additional context**
n/a

deck55, 2023-07-13:
I would think using the Save Variables to disk when you make a mesh to save a date, then check that variable in your start print macro.

[Camman Templates](https://www.klipper3d.org/Command_Templates.html)

[Save_Variables](https://www.klipper3d.org/Config_Reference.html#save_variables)

Ltek, 2023-07-14:
> I would think using the Save Variables to disk when you make a mesh to save a date, then check that variable in your start print macro.
> 
> [Command Templates](https://www.klipper3d.org/Command_Templates.html)
> 
> [Save_Variables](https://www.klipper3d.org/Config_Reference.html#save_variables)

I'm not understanding... how does this turn on or off the Adaptive Mesh, which runs prior to each/all prints?

I'd like to do a 'default' mesh that maps the entire bed... so most of the time it uses that, automatically without doing anything. Then, on the occasion that I want to run Adaptive Mesh (just prior to printing), there is a 'button' (macro) that turns ON Adaptive Mesh but only for the next print.

kyleisah, 2023-07-14:
This is an interesting idea! I wonder if there's a way to make that happen... Off the cuff I can think of one way, but it'll make object exlusion not work for the print it's enabled for.

deck55, 2023-07-14:
My initial thought was to add the save variable to config. Add or adjust a button macro to do a bed mesh, no objects would be loaded so a full bed mesh would be created, in the macro it would then store a date. Then in start macro add a check if today is more than x number of days, if it is use kamp to do a new mesh, if no then just load saved mesh.

Then this would continue to use kamp to create a mesh until you hit the button to update your new full bed mesh.

Not 100% sure if it cam be incorporated into kamp, but I think it could be done.

Ltek, 2023-07-15:
> This is an interesting idea! I wonder if there's a way to make that happen... Off the cuff I can think of one way, but it'll make object exclusion not work for the print it's enabled for.

Thanks. I'm not a coder so its beyond my ability so hoping you can :-)
I think the most universal way would be...

1. code looks to see if the "button" is ON. 
2. If button is "on", KAMP Adaptive Mesh runs
3. If buttons is "off", the code looks for existing 'default' bed mesh profile.
4. if the 'default' profile exists then KAMP Adaptive Mesh doesn't run and 'default' profile is loaded
5. if the 'default' profile does not exist then KAMP Adaptive Mesh runs

kyleisah, 2023-07-15:
> > This is an interesting idea! I wonder if there's a way to make that happen... Off the cuff I can think of one way, but it'll make object exclusion not work for the print it's enabled for.
> 
> 
> 
> Thanks. I'm not a coder so its beyond my ability so hoping you can :-)
> 
> I think the most universal way would be...
> 
> 
> 
> 1. code looks to see if the "KAMP Button" is "Always Run" or "Smart Run". 
> 
> 2. for "Always Run"... KAMP Adaptive Mesh runs every time
> 
> 3. for "Smart Run"... KAMP code looks for existing 'default' bed mesh profile.
> 
> 4. if the 'default' profile exists then KAMP Adaptive Mesh doesn't run and 'default' profile is loaded
> 
> 5. if the 'default' profile does not exist then KAMP Adaptive Mesh runs

I'm in no way a coder either haha, I just have a pretty good understanding of klipper macros lol. What you're asking for should be doable, but it would take some figuring out on my end, and doesn't exactly fit the ethos of KAMP. There are ways to toggle a variable like a button and write that to a saved file so it works across restarts, which is what you would likely need, but I've never messed with that functionality before so I'd have to figure it out.

Ltek, 2023-07-18:
@kyleisah another idea (hopefully easy).  If the print about to happen is the same as the prior print, the bed measurements will not have changed, thus Adaptive Bed Mesh should be skipped. Would you be able to add a check to see if the 'print name' is the same, or if the mesh area is the same... then it skips the bed mesh calibration?

kyleisah, 2023-07-23:
I'm closing this because this is not the ethos of KAMP in my opinion. You are more than welcome to create a fork and attempt something like this yourself, but myself and many others are perfectly fine just using the adaptive mesh functionality and probing prior to each print or recalling a saved mesh whenever it is deemed necessary. What you're asking can be done, but *automating* it is the difficult part.

closed 2023-07-23 by kyleisah