kyleisah/Klipper-Adaptive-Meshing-Purging #121: FEATURE REQUEST - MMU Purge Tower object detection

kyleisah, 2023-07-14:
There's been an interest in getting KAMP to detect purge towers with MMU setups. I don't have an MMU, but there *might* be a way that we can at least get KAMP to label a purge tower object so the correct area is meshed.

Dropping these Prusaslicer placeholders here so I don't forget and I can work on this later:

```gcode
WIPE_TOWER_PARAMS HAS_TOWER=[has_wipe_tower] X_ORIGIN=[wipe_tower_x] Y_ORIGIN=[wipe_tower_y] WIDTH=[wipe_tower_width]
```

Might be able to get KAMP to convert this information into an EXCLUDE_OBJECT_DEFINE command.

scottgrim, 2023-07-14:
From my digging, I don't think you can get one of the dimensions of the wipe tower.  These are all config variables but at the time of slicing, PS determines how "long" the tower needs to be to accommodate the purge volumes.  You'd probably need to reproduce whatever algorithm it uses for that to determine the final dimension of the tower.

kyleisah, 2023-07-14:
@ksummers92 Just tagging you here in case you were interested in following along

ksummers92, 2023-07-14:
Oh man I am. I just had this issue yesterday. Just have to make sure the purge tower is the behind any parts to make sure the purge line doesn’t get run over.

kyleisah, 2023-07-14:
If any of you guys are able to, please add this command:

`WIPE_TOWER_PARAMS HAS_TOWER=[has_wipe_tower] PRIMING=[single_extruder_multi_material_priming] X_ORIGIN=[wipe_tower_x] Y_ORIGIN=[wipe_tower_y] WIDTH=[wipe_tower_width] PRINT_MIN=[first_layer_print_min] PRINT_MAX=[first_layer_print_max]`

above the PRINT_START macro in your start gcode like this:

[image]

Then, slice an MMU print like you usually would and upload the gcode here **as a .txt file**. Thank you!

kyleisah, 2023-07-14:
All this will do is grab these placeholder values from the slicer and output them as something I can easily read. Plus, I don't have an MMU so I don't know what I'm doing in terms of slicing correctly. 😅

scottgrim, 2023-07-14:
WIPE_TOWER_PARAMS HAS_TOWER=1 PRIMING=0 X_ORIGIN=212.28 Y_ORIGIN=209.387 WIDTH=60 PRINT_MIN=135.3 PRINT_MAX=274.108
print_start EXTRUDER=215 BED=60 FIL=PLA TOOL=0

Do you need the entire gcode file or just this?

scottgrim, 2023-07-14:
Also, here's the preview as reference.
[image]

scottgrim, 2023-07-14:
[MM Test print 9 colors.gcode.txt](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12058960/MM.Test.print.9.colors.gcode.txt)

scottgrim, 2023-07-14:
And here's the same output but setting it to only 3 colors which produces a smaller tower.  I guess including blank space is better than not including the tower at all, but it's, of course, sub-optimal.
`WIPE_TOWER_PARAMS HAS_TOWER=1 PRIMING=0 X_ORIGIN=212.28 Y_ORIGIN=209.387 WIDTH=60 PRINT_MIN=135.3 PRINT_MAX=274.108`
[image]

scottgrim, 2023-07-14:
> Oh man I am. I just had this issue yesterday. Just have to make sure the purge tower is the behind any parts to make sure the purge line doesn’t get run over.

That's what I've been doing. It's an ok workaround and I'm in the habit of putting them in the right rear anyway for Octolapse on my other printer.

kyleisah, 2023-07-14:
> WIPE_TOWER_PARAMS HAS_TOWER=1 PRIMING=0 X_ORIGIN=212.28 Y_ORIGIN=209.387 WIDTH=60 PRINT_MIN=135.3 PRINT_MAX=274.108 print_start EXTRUDER=215 BED=60 FIL=PLA TOOL=0
> 
> Do you need the entire gcode file or just this?

Yeah, I need the gcode just so I can work out where everything is happening. I see you uploaded one though, so all good. Appreciate it! 😄 I will do some digging and see if I can make something. Not sure about the eventual implementation into KAMP but we can at least try now and figure that out later. :)

scottgrim, 2023-07-14:
Well if we can figure out the right variables to pass, there's a klipper command to manually add and exclude object which should do the trick.

kyleisah, 2023-07-14:
> 

That's exactly it, gonna fake it to make it. 😆 

I uploaded you gcode to my mainsail instance to make excluded objects process. I had no idea it would still be considered 9 objects, that's pretty cool.

[image]

scottgrim, 2023-07-14:
Those are intentionally broken into 9 objects so that you can just set the extruder to use for each object rather than having to manually paint on color.

kyleisah, 2023-07-16:
I'm gonna ask this before I start really digging into this. 

I can easily make KAMP just do a full bed mesh and skip purging if there's a wipe tower present. 

*OR*

I can do my best to make KAMP generate a wipe block object (that likely won't be perfect every time) and meshing and purging *should* adjust correctly assuming I can get the calculations correctly. 

First option is like a 0 in difficulty for me, second is like a 5/6 in difficulty for me. I'm mostly curious to just see if it can be done good enough to work right. 

So are you guys willing to wait for something that might not be 100%, or do you guys want something that will work every time, but only disable KAMP when a wipe tower is detected?

scottgrim, 2023-07-16:
It's not a big deal since I have a completely functional workaround so don't feel rushed to come up with a solution.
Also, I don't think not purging when a tower is active would work because it generally starts the tower after the first color is finished so that would leave no pre-print purge at all.  It's also not that important to me to have the area where the purge tower prints be probed because I don't care if the purge tower first layer is perfect as long as it adheres which isn't a problem on my pretty flat bed.

ksummers92, 2023-07-16:
I agree with @scottgrim. A suggestion for the purge might be to do a more traditional purge if a tower is detected and place it at an extreme of the build space. That would provide the necessary purge and eliminate any interference issues.

kyleisah, 2023-07-16:
So what I'm hearing is you guys think running a full mesh is fine, but the purge should still happen at an extreme of the bed size so it will most likely be out of the way if the wipe tower. I can probably whip that up pretty quick. Would that work until a better solution is presented?

scottgrim, 2023-07-16:
Depends on what you mean by "full mesh". I still want the mesh to be limited to the print area (even if it excludes the tower) but an option to have it do the purge at like (10,10) if there's a tower would be a good parameter to have.

I actually pass a parameter to my print_start to determine if I want to do adaptive mesh vs normal mesh vs. just use a stored mesh which is pretty reliable since the voron bed doesn't change that much assuming the same heatsoak temps.

kyleisah, 2023-07-16:
> Depends on what you mean by "full mesh". I still want the mesh to be limited to the print area (even if it excludes the tower) but an option to have it do the purge at like (10,10) if there's a [...]

Right, but the problem is, since the wipe tower isn't labeled by the slicer, it sometimes may not be in the bounds of the meshed area. My beds are pretty flat too, but not everyone is as fortunate. Just trying to cover it for as many users as I can.

scottgrim, 2023-07-16:
You know, just adding a variable to SETUP_VORON_PURGE to let us set a static location that skips the dynamic calculation would do the trick. Then in our PRINT_START we can decide if we want to let it do dynamic or if we want to override it.  We can pass our print_start a param indicating if there's a wipe_tower with the has_wipe_tower substitution.

scottgrim, 2023-07-16:
Actually, it may already be there.  ADAPTIVE_ENABLE?

So really, I don't think we need any changes unless/until we can determine the exact coordinates/size of the wipe tower dynamically.

scottgrim, 2023-07-16:
> Right, but the problem is, since the wipe tower isn't labeled by the slicer, it sometimes may not be in the bounds of the meshed area. My beds are pretty flat too, but not everyone is as fortunate. [...]

That's fair.  Just please make it user selectable.  I can't imagine a bed that's warped enough that it would impact a wipe tower which doesn't need to be perfect but I've seen some printers that belonged in a horror story so I suppose it's possible they'd be warped enough to cause adhesion issues.

scottgrim, 2023-07-16:
But just having a toggle to the SETUP command to let people decide in print_start if they want it dynamic/adaptive or not would be best since we can tell our print_start if there's a tower or not.  Leave the control in the user's hands and show examples?

ksummers92, 2023-07-17:
@scottgrim, you may want to update your KAMP config. The SETUP command is no longer utilized in the latest release. The being said, something could be added to the `_KAMP_Settings` macro to enable and disable adaptive purge and mesh. I like the idea of the macro being able to intuit the need from the sliced gcode and allowing the user to select whether the ability to intuit should be applied. For example, have a variable that the user can set to adjust for wipe towers (call it `variable_wipe_tower_compensation` or something). Perhaps instead of resorting to a full bed mesh or the more arduous task of labeling the wipe tower, the mesh margin could be temporarily expanded by some amount to achieve better (but not perfect) coverage? Or, since the wipe tower parameters can be passed, then the mesh could be expanded in the direction without necessarily having to define it as an object.

scottgrim, 2023-07-17:
Hmm.. that sounds like a pretty breaking change. I just went through and let Moonraker update things but haven't tried printing since.  I hope those changes have backward compatibility.

kyleisah, 2023-07-18:
I'm going to say best course of action is to just remove all of KAMP, and install it again. There was a lot of hands in the original implementation of moonraker updating, and it was really frustrating to understand and complicated to work with. Since I'm the only one really providing support since the beginning, I rewrote everything so I could better understand it and provide better support. There are going to be bumps in the road, I'm no professional when it comes to this stuff but I have always done my best to provide something quality.

scottgrim, 2023-07-18:
I get it and I appreciate your efforts.  It's just a little frustrating to get a breaking update without any kind of warning when auto updating is encouraged.

I think I have it sorted.  For reference, I did have to delete KAMP and start from scratch because it didn't install the KAMP_Settings.cfg just updating from Moonraker.

For the record, I always thought the way it worked before was a bit "busted" and this way makes a lot more sense.

kyleisah, 2023-07-18:
That was exactly the reason I worked on this, it felt very rushed and janky in the past, and now it's easy for me to implement features without breaking anything (fingers crossed). As klipper/moonraker evolves I'm hoping there will be way I can push notifications to instances of klipper/moonraker to make people more aware that something needs more attention before they start printing, but for now I'll just have to do my best to not break any more printers. 😅

kyleisah, 2023-08-20:
This is no longer in my hands. People who want this feature will need to create github issues for their respective slicer.

closed 2023-08-20 by kyleisah