kyleisah/Klipper-Adaptive-Meshing-Purging #123: Adaptive Purge/Line Purge - Purging inside of print area

Serox13, 2023-07-16:
**Describe the issue**
Adaptive purge line is creating a line in the middle of the print area, not on top 

**What slicer are you using?**
SuperSlicer

**Troubleshooting**
What troubleshooting steps have you tried?
i have moved the variable_purge_margin: from 10(default) to 20 still happens
i had 4 prints where this didn't happen purge line was outside of the printed area
i have tried the VORON_PURGE instead of LINE_PURGE still happens

i reinstalled everything from scratch and I have reloaded fresh unedited cfg files, the issue still remains.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Klipper log file:**
Please add klipper log file here...
[klippy (7).log](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12063425/klippy.7.log)

**Sliced Gcode**
Please add the sliced gcode file you are using here...
[gcode.txt](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12063436/gcode.txt)

**Additional context**
Add any other context about the problem here.
happens no matter the size of the printed area
adaptive meshing works perfectly it engages the printed area to mesh

i am new to 3d printing and gcode coding, hence the simple questions. Once things are explained i can understand them
all files have been uploaded thanks for the help

kyleisah, 2023-07-16:
Hello! 

That's very odd. Haven't seen that happen yet. I'm away from a computer that I could preview your gcode with and see what's going on, but I noticed that the object the slicer is loading has a ton of polygon points. I also see there's some support material gcode. From what I can see with your config, everything looks right. 

The voron purge purges at the corner of tue print, where the line purges at the front center, unless there's no room in front of it, then it'll move to the side. 

I'm curious if your support material isn't being labeled by your slicer as an object Klipper can cancel and that's why the purge is ending up in the wrong location. 

I am a little busy today, but I will take a look the next chance I get.

kyleisah, 2023-07-16:
As a stop gap, uncomment the line purge from your print_start and increase the amount of skirt loops in your slicer profile. That way you can still purge (sort of) before the actual print begins.

kyleisah, 2023-07-18:
Ah, okay, I see what's going on here. So it looks like you had a large brim on this print because it's a test tower. KAMP did exactly what it should do. Only problem is, almost all slicers for some reason do not define brims or skirts as objects. This is problematic when you need a larger purge margin than usual. I've been trying to come up with ways to make the purges adapt to these things, but honestly the easiest solution would be to either set your brim width less than the purge margin or vice versa, or somehow convince the software owner to make their slicer label these objects. There's nothing I can do for you for this issue unfortunately, we all will just have to wait until these features start to become normal.

Thank you for reaching out!

closed 2023-07-18 by kyleisah