kyleisah/Klipper-Adaptive-Meshing-Purging #124: Mesh fails on thin long parts

MartijnGevaert, 2023-07-16:
**Describe the issue**
The meshing fails on thin long parts. I have a part that is 0.9mm wide (2 layer lines). The meshing fails for this with following message:

bed_mesh: min/max points too close together

**What slicer are you using?**
Prusaslicer

**Troubleshooting**
What troubleshooting steps have you tried?

**Screenshots**
part:
[image]
klipper console:
[image]

**Sliced Gcode**
Please add sliced gcode file you are using here...  -> Can't do, Github does not seem to allow this.

kyleisah, 2023-07-16:
I think you might be using an old version of KAMP but that's not really related here. 

Were you testing the limits of KAMP or something? I purge more material than that 😂

kyleisah, 2023-07-16:
This has actually come up a time or two before, I've been meaning to figure this out.

kyleisah, 2023-07-16:
If you update to the newest KAMP, you can set a `mesh_margin` which will expand the object bounds by a set distance which should allow you to print objects as small as this.

MartijnGevaert, 2023-07-16:
I was actually trying to replicate a bug I came across in Klipper. Hence the extreme geometry of the part.
I'll make sure to set up my mesh margin

kyleisah, 2023-07-18:
Thanks for reaching out! I think you would be best just having a margin of 5 or more so even for these tiny parts, KAMP should be able to adapt.

closed 2023-07-18 by kyleisah