kyleisah/Klipper-Adaptive-Meshing-Purging #125: Toolhead moves to front of printer from "smart park" position prior to voron purge

SuperDodge, 2023-07-17:
**Describe the issue**
I am using the "smart park" feature to park the toolhead prior to final heatup but then the toolhead moves all the way to the front of the printer and then back to the same location when ready to purge.

**What slicer are you using?**
SuperSlicer

kyleisah, 2023-07-18:
There's some difficulty here because the purges have different origins. The voron purge is at the corner of the print, and the line purge is directly in front. There may be some way to make the park go to the actual purge start location, but that would be broken if the user has both purges on their config. It's not really that big of a deal in my opinion. It's only saving you a second or two of travel from a typical parking location. Thank you for reaching out!

kyleisah, 2023-07-18:
Ah, sorry, I had clicked the wrong issue and wrote my reply. Can you upload a klipper log?

kyleisah, 2023-07-22:
I'm closing this issue because no log was provided and I can't reproduce it.

closed 2023-07-22 by kyleisah

SuperDodge, 2023-07-23:
Log attached (cleaned has stats lines removed for clarity, klippy is first half of full log trimmed to meet GitHub file size limits)

[cleaned.log](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12140174/cleaned.log)
[cleaned.log](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12140186/cleaned.log)

kyleisah, 2023-07-24:
You're calling a park position in your print_start as well as using smart_park. If you're using the "better print start macro" by jontek, you'll need to comment out any moves to x_wait and y_wait and just use smart_park instead.