kyleisah/Klipper-Adaptive-Meshing-Purging #126: Have Smart Park position at same X/Y coordinates that Purge starts

ChaosBlades, 2023-07-17:
I like to have my Smart Park at 0.4 mm so while the extruder is heating it does not ooz a big string. If I do this the Smart Park will sometimes leave this ooz dot inside my print. Please have Smart Park position the nozzle in the same X/Y coordinates as the start of Purge. Then it will just need to move Z up or down and start purging.

kyleisah, 2023-07-18:
There's some difficulty here because the purges have different origins. The voron purge is at the corner of the print, and the line purge is directly in front. There may be some way to make the park go to the actual purge start location, but that would be broken if the user has both purges enabled in their config. It's not really that big of a deal in my opinion. It's only saving you a second or two of travel from a typical parking location. Thank you for reaching out!

kyleisah, 2023-07-18:
I can maybe have the smart park origin account for the margin that's defined for purging. That could possibly solve your issue!

ChaosBlades, 2023-07-18:
It is still leaving a small dot of filament on the print bed within the print area.

kyleisah, 2023-07-18:
> It is still leaving a small dot of filament on the print bed within the print area.

I understand how that can be an issue now, and is likely oversight on my part.

If you're willing to try, please change your `Smart_Park` macro to this:

```yaml
[gcode_macro Smart_Park]
description: Parks your printhead near the print area for pre-print hotend heating.
gcode:

    {% set z_height = printer["gcode_macro _KAMP_Settings"].smart_park_height | float %}                                            # Pull park height value from _KAMP_Settings
    {% set purge_margin =  printer['gcode_macro _KAMP_Settings'].purge_margin | float %}                                            # Pull purge margin value from _KAMP_Settings
    {% set center_x = printer.toolhead.axis_maximum.x / 2 | float %}                                                                # Create center point of x for fallback
    {% set center_y = printer.toolhead.axis_maximum.y / 2 | float %}                                                                # Create center point of y for fallback
    {% set all_points = printer.exclude_object.objects | map(attribute='polygon') | sum(start=[]) %}                                # Gather all object points
    {% set x_min = all_points | map(attribute=0) | min | default(center_x) %}                                                       # Set x_min from smallest object x point
    {% set y_min = all_points | map(attribute=1) | min | default(center_y) %}                                                       # Set y_min from smallest object y point
    {% set travel_speed = (printer.toolhead.max_velocity) * 60 | float %}                                                           # Set travel speed from config

    {% if printer['gcode_macro _KAMP_Settings'].purge_margin > 0 and x_min != center_x and y_min != center_y %}
        {% set x_min = x_min - purge_margin %}
        {% set y_min = y_min - purge_margin %}
    {% endif %}

    G0 X{x_min} Y{y_min} F{travel_speed}                                                                                            # Move near object area
    G0 Z{z_height}                                                                                                                  # Move Z to park height
```

kyleisah, 2023-07-18:
@ChaosBlades have you had a chance to try this fix?

ChaosBlades, 2023-07-19:
@kyleisah That is perfect. It is on the same Y position as the purge line.

kyleisah, 2023-07-19:
Awesome, good to hear. That change will be pushed in the next update.

kyleisah, 2023-07-20:
Just a word of warning because I just had this happen to me with the adjusted macro, if a plate is too full, the park will fail. I'll have to see how I can fix that.

kyleisah, 2023-07-20:
```yaml
[gcode_macro Smart_Park]
description: Parks your printhead near the print area for pre-print hotend heating.
gcode:

    {% set kamp_settings = printer["gcode_macro _KAMP_Settings"] %}                                                                 # Pull all variables from _KAMP_Settings
    {% set z_height = kamp_settings.smart_park_height | float %}                                                                    # Set Z height variable
    {% set purge_margin = kamp_settings.purge_margin | float %}                                                                     # Set purge margin variable
    {% set verbose_enable = kamp_settings.verbose_enable | abs %}                                                                   # Set verbosity
    {% set center_x = printer.toolhead.axis_maximum.x / 2 | float %}                                                                # Create center point of x for fallback
    {% set center_y = printer.toolhead.axis_maximum.y / 2 | float %}                                                                # Create center point of y for fallback
    {% set all_points = printer.exclude_object.objects | map(attribute='polygon') | sum(start=[]) %}                                # Gather all object points
    {% set x_min = all_points | map(attribute=0) | min | default(center_x) %}                                                       # Set x_min from smallest object x point
    {% set y_min = all_points | map(attribute=1) | min | default(center_y) %}                                                       # Set y_min from smallest object y point
    {% set travel_speed = (printer.toolhead.max_velocity) * 60 | float %}                                                           # Set travel speed from config

    {% if purge_margin > 0 and x_min != center_x and y_min != center_y %}                                                           # If objects are detected and purge margin 
        {% set x_min = [ x_min - purge_margin , x_min ] | max %}                                                                    # value is greater than 0, move
        {% set y_min = [ y_min - purge_margin , y_min ] | max %}                                                                    # to purge location + margin
    {% endif %}

                                                                                                                                    # Verbose park location
    {% if verbose_enable == True %}

    { action_respond_info("Smart Park location: {},{}.".format(
        (x_min),
        (y_min),
    )) }

    {% endif %}

    G0 X{x_min} Y{y_min} F{travel_speed}                                                                                            # Move near object area
    G0 Z{z_height}                                                                                                                  # Move Z to park height 
```

This is the proposed fix.

kyleisah, 2023-07-21:
@kieraneglin the proposed fix is up above, and seems to work pretty well from my testing.

kyleisah, 2023-08-20:
This is now in the main release. Please update!

closed 2023-08-20 by kyleisah