kyleisah/Klipper-Adaptive-Meshing-Purging #129: Adaptive mesh ignoring my max and min settings

karlc28, 2023-07-22:
**Describe the issue**
As my probe cannot reach 100% of my bed I have set max and min settings in klipper as normal but when trying to print things nearly as big as my bed it ignores the max and min I have set then tries to move beyond limits which makes the print fail

**What slicer are you using?**
Prusaslicer,

**Troubleshooting**
Changing the max and min 

**Screenshots**
[image]
My max x in mesh is 220 and y is 200

kyleisah, 2023-07-22:
Can you upload a klipper log file?

kyleisah, 2023-07-22:
I think you might have to account for your probe offsets when defining the mesh max and min. 

For example, if your probe offsets for X and Y are (20,20) and you want a mesh min of (20,20), you have to adjust the values for your probe offset as well, so you would do (40,40).

karlc28, 2023-07-23:
My Probe offset was accounted for so my max x mesh is 220 and probe offset is -21 so it shouldn't move past 241 but it seems to be ignoring that

kyleisah, 2023-07-23:
@karlc28 please upload a klipper log so I can better assist you with this

karlc28, 2023-07-23:
[klippy.log](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12138525/klippy.log)

kyleisah, 2023-07-23:
[image]

Keep in mind this is how Klipper interprets probe offset.

kyleisah, 2023-07-24:
Looking at your offsets, you've calculated your mesh bounds in your config incorrectly. You need to account for your probe offsets for your mesh_min and mesh_max. 

Your BLTouch X offset is `-30.5`, so if the nozzle is at 0,0, the probe tip will be at `-30.5,-29`. So your bed mesh min should be `30.5, 29` but for the sake of simplicity, I recommend using these values:

```yaml
[bed_mesh]
.
.
mesh_min: 35,35
mesh_max: 220,220
```

Or something like this. With your probe having a negative offset, you need to ADD to your mesh min to compensate, as well as obey the maximum bounds your nozzle can be at.

kyleisah, 2023-07-24:
I'm not 100% sure on this, but to get the absolute most out of your bed mesh bounds, you'd need something like this:

```yaml
[bed_mesh]
.
.
mesh_min: 30.5,29 # to get the probe tip to be at 0,0 of the bed
mesh_max: 260,230.5 # to get the probe tip closest to XY max as possible while still obeying machine boundaries
```

More information can be found here: https://www.klipper3d.org/Bed_Mesh.html#basic-configuration

The bed_mesh min and max points are called relative to the probe's location.

I could be wrong on this, I don't use a probe with offsets, I probe with the nozzle, but this is my very basic understanding.

karlc28, 2023-07-27:
my mesh max is 220,200
my bl touch offset is x: -30.5 y:-29
so the adaptive mesh should never try to push my nozzle past 250.5, 229
But as you can see here it is trying to push Y past 235 which makes no sense 
when i turn adaptive meshing off and have standard klipper mesh it obeys the min and max [rule]
[image]

here is my klippy.log

[klippy (1).log](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12182514/klippy.1.log)

karlc28, 2023-07-27:
[image]
as shown here it sets Adapted mesh bounds: (29.123, 22.809), (205.969, 206.196).
the 206.196 is greater than the 200 i have set in my bed mesh max

karlc28, 2023-07-27:
found the issue i had expanded mesh by 10mm in KAMP settings and didnt realise this would push it past the limits set in printer.cfg i believe this is now resolved

closed 2023-07-27 by karlc28