kyleisah/Klipper-Adaptive-Meshing-Purging #130: Adaptive Mesh ignores Z-Offset?

MaffooClock, 2023-07-22:
**Describe the issue**
I don't know that this is a bug in this plugin, especially since no one else has reported anything even similar, but...

I have a well-tuned Voron 2.4 (in use for several months now) with a [Klicky probe](https://github.com/jlas1/Klicky-Probe) and the [Automatic Z Calibration plugin](https://github.com/protoloft/klipper_z_calibration).  Before using KAMP, I used the "default" bed mesh (performed manually and saved, not as part of the START_PRINT), and my prime/purge method was just to draw a line down the left side.

After Implementing this plugin, I took a look at the [A better print_start macro](https://github.com/jontek2/A-better-print_start-macro) linked in at the bottom of the README, and noted the sequence of G28, QUAD_GANTRY_LEVEL, G28 Z, CALIBRATE_Z, then BED_MESH_CALIBRATE, followed by SMART_PARK to wait for the nozzle to reach temperature before VORON_PURGE.  I didn't just copy/paste jontek2's macro, I just used it as a guide to ensure mine followed that same sequence.

During the beginning of a print, everything seemed to work normally: the auto Z-calibration executed as it always did (Fluidd showed ~0.576mm z-offset, which is about normal), then the adaptive bed mesh did it's thing and seemed to work just fine, and finally the toolhead parked where I figured it would.  But when VORON_PURGE started, the toolhead was jammed all the way town into the build plate and gouged a _HUGE_ canyon into my build plate.

For attempt 2, I disabled (commented out) the BED_MESH_CALIBRATE in my START_PRINT macro and left everything else the same.  The print executed just fine, including the nifty little Voron logo for the purge.

For attempt 3, after triple-checking that I had just the right sequence of event in my START_PRINT, I re-enabled BED_MESH_CALIBRATE.  Just like the last two times, everything worked perfectly, except the toolhead was once again crashed deep into the build plate during VORON_PURGE after BED_MESH_CALIBRATE.

**What slicer are you using?**
OrcaSlicer

**Troubleshooting**
See above.

**Screenshots**
N/A

**Klipper log file:**
I copied this from the console in Fluidd, which shows what was logged during the sequence described in the second paragraph above (G28, QUAD_GANTRY_LEVEL, G28 Z, CALIBRATE_Z, BED_MESH_CALIBRATE, SMART_PARK, VORON_PURGE, \*crash\*).  Attached is [klippy.log](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12136781/klippy.log), too.
```
09:14:34 $ SDCARD_PRINT_FILE FILENAME="a_thing_to_print.gcode"
09:14:34 File opened:a_thing_to_print.gcode Size:3318712
09:14:34 File selected
09:16:06 // moving to a safe Z distance
09:16:06 // Homing Z
09:16:08 // probe: TRIGGERED
09:16:08 // Probe already docked
09:16:09 // probe: TRIGGERED
09:16:16 // probe: TRIGGERED
09:16:16 // Attaching Probe
09:16:19 // probe: open
09:16:25 // probe at 214.000,342.000 is z=0.135625
09:16:27 // probe at 214.000,342.000 is z=0.124375
09:16:29 // probe at 214.000,342.000 is z=0.115000
09:16:30 // probe at 214.000,342.000 is z=0.106250
09:16:30 // Probe samples exceed tolerance. Retrying...
09:16:32 // probe at 214.000,342.000 is z=0.099375
09:16:33 // probe at 214.000,342.000 is z=0.093125
09:16:35 // probe at 214.000,342.000 is z=0.085000
09:16:35 // Probe samples exceed tolerance. Retrying...
09:16:36 // probe at 214.000,342.000 is z=0.080000
09:16:38 // probe at 214.000,342.000 is z=0.077500
09:16:39 // probe at 214.000,342.000 is z=0.073125
09:16:42 // probe at 205.000,323.000 is z=3.195625
09:16:43 // probe at 205.000,323.000 is z=3.206875
09:16:45 // probe at 205.000,323.000 is z=3.216875
09:16:46 // probe at 205.000,323.000 is z=3.218125
09:16:46 // Probe samples exceed tolerance. Retrying...
09:16:48 // probe at 205.000,323.000 is z=3.213750
09:16:49 // probe at 205.000,323.000 is z=3.216250
09:16:51 // probe at 205.000,323.000 is z=3.216875
09:16:56 // probe at 175.000,155.250 is z=4.108750
09:16:58 // probe at 175.000,155.250 is z=4.115000
09:16:59 // probe at 175.000,155.250 is z=4.115625
09:17:01 // probe at 175.000,155.250 is z=4.115625
09:17:01 // Z-CALIBRATION: probe=4.116 - (switch=3.216 - nozzle=0.077 + switch_offset=0.400) --> new offset=0.576875
09:17:01 // HINT: z position_endstop=0.100 - offset=0.576875 --> possible z position_endstop=-0.477
09:17:01 // Docking Probe
09:17:07 // probe: TRIGGERED
09:17:07 // Algorithm: lagrange.
09:17:07 // Default probe count: 5,5.
09:17:07 // Adapted probe count: 3,3.
09:17:07 // Default mesh bounds: (40.0, 40.0), (310.0, 310.0).
09:17:07 // Mesh margin is 0, margin not increased.
09:17:07 // Fuzz amount is 0, mesh points not fuzzed.
09:17:07 // Adapted mesh bounds: (130.0, 127.0), (220.0, 223.0).
09:17:07 // KAMP adjustments successful. Happy KAMPing!
09:17:07 // Attaching Probe
09:17:10 // probe: open
09:17:10 // Generating new points...
09:17:10 // bed_mesh: generated points
         // Index | Tool Adjusted | Probe
09:17:10 // 0 | (130.0, 107.2) | (130.0, 127.0)
09:17:10 // 1 | (175.0, 107.2) | (175.0, 127.0)
09:17:10 // 2 | (220.0, 107.2) | (220.0, 127.0)
09:17:10 // 3 | (220.0, 155.2) | (220.0, 175.0)
09:17:10 // 4 | (175.0, 155.2) | (175.0, 175.0)
09:17:10 // 5 | (130.0, 155.2) | (130.0, 175.0)
09:17:10 // 6 | (130.0, 203.2) | (130.0, 223.0)
09:17:10 // 7 | (175.0, 203.2) | (175.0, 223.0)
09:17:10 // 8 | (220.0, 203.2) | (220.0, 223.0)
09:17:13 // probe at 130.000,107.250 is z=4.149375
09:17:14 // probe at 130.000,107.250 is z=4.156875
09:17:14 // probe at 130.000,107.250 is z=4.159375
09:17:16 // probe at 175.000,107.250 is z=4.125000
09:17:17 // probe at 175.000,107.250 is z=4.126875
09:17:18 // probe at 175.000,107.250 is z=4.126875
09:17:20 // probe at 220.000,107.250 is z=4.138750
09:17:21 // probe at 220.000,107.250 is z=4.141250
09:17:22 // probe at 220.000,107.250 is z=4.141875
09:17:24 // probe at 220.000,155.250 is z=4.130000
09:17:25 // probe at 220.000,155.250 is z=4.133125
09:17:26 // probe at 220.000,155.250 is z=4.135000
09:17:28 // probe at 175.000,155.250 is z=4.126250
09:17:29 // probe at 175.000,155.250 is z=4.128125
09:17:30 // probe at 175.000,155.250 is z=4.128750
09:17:32 // probe at 130.000,155.250 is z=4.128125
09:17:33 // probe at 130.000,155.250 is z=4.131250
09:17:34 // probe at 130.000,155.250 is z=4.133125
09:17:36 // probe at 130.000,203.250 is z=4.118125
09:17:37 // probe at 130.000,203.250 is z=4.119375
09:17:38 // probe at 130.000,203.250 is z=4.120000
09:17:40 // probe at 175.000,203.250 is z=4.124375
09:17:41 // probe at 175.000,203.250 is z=4.124375
09:17:43 // probe at 220.000,203.250 is z=4.148750
09:17:44 // probe at 220.000,203.250 is z=4.150625
09:17:45 // probe at 220.000,203.250 is z=4.150000
09:17:45 // Mesh Bed Leveling Complete
09:17:45 // Bed Mesh state has been saved to profile [default]
         // for the current session. The SAVE_CONFIG command will
         // update the printer config file and restart the printer.
09:17:45 // Docking Probe
09:17:52 // probe: TRIGGERED
09:18:29 // Moving filament tip 0.0mms
09:18:29 // KAMP purge is using firmware retraction.
09:18:40 // Klipper state: Shutdown
09:19:50 $ FIRMWARE_RESTART
09:19:50 // Klipper state: Disconnect
```

**Sliced Gcode**
N/A

**Additional context**
I've reviewed the BED_MESH_CALIBRATE macro in [Adaptive_Meshing.cfg](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/blob/23bee5be0be1c88536cd91a6f203585918566e33/Configuration/Adaptive_Meshing.cfg), and it doesn't appear to be _doing_ anything to the offset that I could see, but the fact remains: after this macro runs, toolhead crashes into build plate even though there is a valid z offset (same value after the meshing as before).  Maybe the problem is in upstream _BED_MESH_CALIBRATE?

kyleisah, 2023-07-22:
I don't know why this would be happening. KAMP doesn't change your Z offset, it just adapts the mesh size, and the purges happen at a configured height. 

Are you using a relative reference index or a zero reference point?

kyleisah, 2023-07-23:
I think what's going on is you're not using a zero_reference_point in your [bed_mesh] config. 

`zero_reference_point: 175,175`

You are calibrating your Z offset to the center of the bed, but not scaling the mesh from the center of your bed. 

Since you have both a probe *and* a Z endstop, you should be using a zero reference point, formerly relative_reference_index. 

It could also likely be something happening from the calibrate_z module, but I don't know enough about it to help. 

Try adding that config to your bed_mesh section and run it again.

jusii, 2023-07-29:
Something similar just happened to me. Voron 2.4 R2 350. After updating all, nozzle crashed to bed. I had my relative_reference_index set to far right corner (24 with 5x5). I guess with this change to zero_reference_point and me not configuring it was the reason. So offset was set according to that corner, but after update it was applied referencing some other point.

kyleisah, 2023-07-29:
> 

So using the zero_reference_point fixed this issue?

jusii, 2023-07-29:
> > 
> 
> So using the zero_reference_point fixed this issue?

For me it did, old z offset was spot on after I added zero_reference_point.

MaffooClock, 2023-07-30:
Hmm... when I add zero_reference_point to [bed_mesh], I am met with a Klipper error: "Option 'zero_reference_point' is not valid in section 'bed_mesh'.  Also, I don't see zero_reference_point in [the documentation](https://www.klipper3d.org/Bed_Mesh.html), so that makes sense.  In fact, Googling "klipper zero_reference_point" turns up basically nothing.  What am I missing?

MaffooClock, 2023-07-30:
Ahh, it's [zero_reference_position](https://www.klipper3d.org/Bed_Mesh.html#configuring-the-zero-reference-position), not zero_reference_point.

MaffooClock, 2023-07-30:
🎉 Confirmed: toolhead no longer crashes into the bed post-meshing after adding zero_reference_position to [bed_mesh].

So I'm betting I would have had the same problem without this plugin if I were performing a BED_MESH_CALIBRATE as part of START_PRINT.  But because I only ever ran that calibration manually and just used the stored mesh, I was oblivious.

@kyleisah: Even though it's completely not your responsibility, I'd like to suggest that you maybe add a short warning about this so that other unwitting users who also use the [Automatic Z Calibration plugin](https://github.com/protoloft/klipper_z_calibration) might be saved from figuring this out the hard way like me.

Thank you for this nifty plugin, and for the help.

MaffooClock, 2023-07-30:
🎉 Confirmed: toolhead no longer crashes into the bed post-meshing after adding zero_reference_position to [bed_mesh].

So I'm betting I would have had the same problem without this plugin if I were performing a BED_MESH_CALIBRATE as part of START_PRINT.  But because I only ever ran that calibration manually and just used the stored mesh, I was oblivious.

So I'm guessing this only effects people who also use the [Automatic Z Calibration plugin](https://github.com/protoloft/klipper_z_calibration).  @kyleisah: even though it's completely not your responsibility, I'd like to suggest that you maybe add a short warning about this so that other unwitting users might be saved from figuring this out the hard way like I had to.

Thank you for this nifty plugin, and for the help.

closed 2023-07-30 by MaffooClock