kyleisah/Klipper-Adaptive-Meshing-Purging #131: Only run Adaptive Mesh if next print is different

Ltek, 2023-07-22:
**Describe the issue**
Only run Adaptive Mesh if the next print (and thus the mesh) will be different.
This will save tons of time and provide the same accuracy

Implementation thoughts...  potentially use the mesh calculation and/or the prior print name to determine if the mesh needs to be updated. If not changed then Adaptive Bed Mesh would be skipped. 

**What slicer are you using?**
Cura & Orcaslicer

**Troubleshooting**
n/a

**Screenshots**
n/a

**Klipper log file:**
n/a

**Sliced Gcode**
n/a

**Additional context**
n/a

kyleisah, 2023-07-23:
You've already made an issue for this over at #120. 

I will close this so we can keep the conversation to one issue.

closed 2023-07-23 by kyleisah

Ltek, 2023-07-23:
> You've already made an issue for this over at #120.
> 
> I will close this so we can keep the conversation to one issue.

This is a very different feature request... I did mention this in that other thread (so I'll go remove that) but this feature is completely different.

this request :  When Adaptive Mesh is ON its 'smart', thus only runs when it needs to
other request : button to turn Adaptive Mesh on/off (separate from 'smart' feature in this thread)