kyleisah/Klipper-Adaptive-Meshing-Purging #132: Support of Simplify3d

Elchschmied, 2023-07-23:
**Describe the issue**
Simplify3d is no longer supported in this new version.

**What slicer are you using?**
Simplify3D
This is a feature request to use KAMP in S3D for the future again. 
Thank you

kyleisah, 2023-07-23:
You can still use the fix that was proposed in the previous versions.

kyleisah, 2023-07-23:
```yaml
# This macro will allow stop-gap functionality of KAMP with Simplify3D. Please add the following to Simplify3D's 
# Start-Gcode section *before* PRINT_START is called, like so:
# .
# .
# MESH_CONFIG X0=[build_min_x] Y0=[build_min_y] X1=[build_max_x] Y1=[build_max_y]
# .
# PRINT_START EXTRUDER_TEMP=[first_layer_temperature] BED_TEMP=[first_layer_bed_temperature]
# .

[gcode_macro MESH_CONFIG]
description: Stop-gap macro for converting Simplify3D's print area into an object for exclude_object
gcode:
    {% set X0 = params.X0 | float %}                # Print Area X Min
    {% set X1 = params.X1 | float %}                # Print Area X Max
    {% set Y0 = params.Y0 | float %}                # Print Area Y Min
    {% set Y1 = params.Y1 | float %}                # Print Area Y Max
    {% set center_x = ((X0 + X1) / 2) | float %}    # Print Area X Center
    {% set center_y = ((Y0 + Y1) / 2) | float %}    # Print Area Y Center

    EXCLUDE_OBJECT_DEFINE NAME=s3d_build_volume CENTER={center_x},{center_y} POLYGON=[[{X0},{Y0}],[{X0},{Y1}],[{X1},{Y1}],[{X1},{Y0}]]
```

Just add this macro somewhere in your configuration and call it **before** PRINT_START, and add the lines commented above to your slicer. This was merely a stop-gap.

closed 2023-07-23 by kyleisah

Elchschmied, 2023-07-23:
Thank you very much