kyleisah/Klipper-Adaptive-Meshing-Purging #133: line_purge generates "Move out of range" for a large print

Eliminateur, 2023-07-23:
**Describe the issue**
With line_purge macro enabled in my start_print macro and a big diagonal print, the macro is not checking or validating the bed boundaries and not adapting accordingly to a clear position, triggering a "**Move out of range**" abort in Klipper.
line_purge is set as:
```
variable_adaptive_enable: True      # Change to False if you'd like the purge to be in the same spot every print
variable_z_height: 0.25              # Height above the bed to purge
variable_purge_amount: 20           # Amount of filament in millimeters to purge
variable_line_length: 50            # Overall desired length of purge line in millimeters, around 1/5th of X axis length is a good starting value
variable_flow_rate: 10              # Desired flow rate in mm3/s (Around 12 for standard flow hotends, around 24 for high flow hotends)
variable_x_default: 10              # Default X location to purge. If adaptive_enable is True, this is overwritten
variable_y_default: 10              # Default Y location to purge. If adaptive_enable is True, this is overwritten
variable_distance_to_object_y: 15   # Y distance in millimeters away from the print area for purging. Must be less than or equal to y_default if adaptive_enable is False
```

**What slicer are you using?**
Orcaslicer 1.6.2

**Troubleshooting**
Commenting line_purge on the startup macro solves the issue

**Screenshots**
this is the sliced object:
[image]

**Klipper log file:**
[klippy.log](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12140144/klippy.log)
line 2185 has the error

**Sliced Gcode**
zipped gcode file
[gcode file.zip](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12140126/gcode.file.zip)

**Additional context**
Add any other context about the problem here.

kyleisah, 2023-07-24:
You're using an old version of KAMP. The newest version has fallbacks for this kind of predicament. I recommend removing all your current KAMP configs and go through the readme again. You'll be glad you did. 😄

pireybr, 2023-07-25:
I'm having the same problem, and I have a updated KAMP, v1.1.1-1-g23bee5b

[1c_King_Head3_GREY-PIF Ellis 45°-ABS 3d LAB.zip](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12156578/1c_King_Head3_GREY-PIF.Ellis.45.-ABS.3d.LAB.zip)

kyleisah, 2023-08-20:
I don't know why that's happening. You likely have a bad config or some kind of gcode offset being imposed. the line purge, when moved past the max position, will move the very edge of the bed.

closed 2023-08-20 by kyleisah