kyleisah/Klipper-Adaptive-Meshing-Purging #134: "Stepper to far in the past" Error message when trying to print Voron_Purge

VORONENTHUSIAST, 2023-07-23:
**Describe the issue**
MCU 'mcu' shutdown: Stepper too far in past
This generally occurs when the micro-controller has been
requested to step at a rate higher than it is capable of
obtaining.

**What slicer are you using?**
Prusaslicer

**Troubleshooting**
Changing the travel_speed variable

**Klipper log file:**
[klippy.log](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12140234/klippy.log)

**Sliced Gcode**
[Z_Tensioner_Bracket_2.txt](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12140239/Z_Tensioner_Bracket_2.txt)

**Additional context**
Hello!! I'm really hoping you can help me out, so this started yesterday, I had to reinstall my printer after some issues, and used the new KAMP setup that's been uploaded to the git.

With the old setup, I had a problem with the Voron_Purge.cfg stalling my motors out so they wouldn't complete the full travel to where the purge should be - I stopped this by putting in my own F values for the move to purge poisition and purge logo lines section of the gcode in the macro, all good!

After installing the new one, I keep getting the above error, I can't seem to get rid of it, not even by using my messy hack from the old install. I'm positive it is to do with the variables and my config causing movements that would be to fast for the drivers, but as I don't know much about Klipper, I could be completely wrong!! :P 

Could you please have a look at my klippy file, I have attached the sliced gcode it's happening with too. I am quite happy going into the Voron_Purge,cfg with winSCP so I can edit it in the correct directory, so if you spot anything please let me know!! 🗡️ 

Thanks in advance!

- S

kyleisah, 2023-07-24:
KAMP uses your max_velocity from the [printer] section to create its travel speeds. Sounds like you might have that too high.

kyleisah, 2023-07-24:
Your X and Y microsteps are pretty high too. You could set those to like 64 and that would probably do it.

VORONENTHUSIAST, 2023-07-24:
Sorry, can you please explain why instead of just telling me to change my config? I have no issues with the microstep settings or my velocity settings when printing, only when trying to use your macro

VORONENTHUSIAST, 2023-07-24:
To let you know, it's working with line_purge - I have halfed the microsteps for the minute, and will try Voron_Purge tomorrow, could you give an explanation as to why this has affected it, maybe? :)

kyleisah, 2023-07-24:
I've got no clue, but I know that having higher microsteps absolutely ruins your motor torque but gives you more resolution. I can't tell you *why* klipper is giving you an error because of it, but I saw that it said "stepper too far in the past" and your XY motors microsteps are at an extreme resolution.

Glad to hear it's working though.

closed 2023-07-24 by kyleisah

VORONENTHUSIAST, 2023-07-24:
Yeah, I did some more reading last night (UK here, couldn't sleep - probably not a good idea to be playing with the printer but hey!) and learned it's trying to do exponentially more steps per mm due to the micro stepping being so high, and of course the torque that gets lost - I'm guessing that and the combo of acceleration caused them to crap out? Thank you for trying to explain a bit - sorry I wasn't trying to be funny - I just like to learn as I'm doing things! :D Thanks for your responses!

VORONENTHUSIAST, 2023-07-24:
Just to let you know - Voron_Purge is a no go, I'm not turning down any acceleration settings as I made a voron because, well voron! Hahaha!

I have noticed that there are calculations in the variables for the purge_line_speed and travel_speed - if you could perhaps explain them, then I can have a play and try to get it working! :D