kyleisah/Klipper-Adaptive-Meshing-Purging #135: variable_mesh_margin should not go beyond bed

jagerman, 2023-07-24:
When printing a nearly-full-bed object with `variable_mesh_margin: 10` I got an erroneous attempt to move to coordinate -2.

For reference I have:

```ini
[probe]
#...
x_offset: 1
y_offset: 22

[bed_mesh]
#...
mesh_min: 10, 30
mesh_max: 290, 290
```
as my probe sits behind the nozzle (and slightly to the right), and a BED_MESH_CALIBRATE runs successfully to probe the full bed (within the min/max limits set above).

But when starting a nearly-full-Y-bed print just now I got this in klippy.log leading up to the move failure:
```
Default mesh bounds: (10.0, 30.0), (290.0, 290.0).
Mesh margin is 10.0, mesh bounds extended by 10.0mm.
Fuzz amount is 0, mesh points not fuzzed.
Adapted mesh bounds: (12.2506, 20.0), (287.749, 288.905).
KAMP adjustments successful. Happy KAMPing!
...
Generating new points...
bed_mesh: generated points
Index |  Tool Adjusted  |   Probe
  0   | (11.3, -2.0)    | (12.3, 20.0)
  1   | (57.2, -2.0)    | (58.2, 20.0)
  2   | (103.1, -2.0)   | (104.1, 20.0)
  ...
```

I think the mesh limits should be taken into account so that a configured mesh margin will be used *if possible*, but will never expand into impossible-to-probe areas of the bed.

kyleisah, 2023-07-24:
I swear I had that working correctly, but I will take another look.

kyleisah, 2023-07-24:
Yeah, looks like that can still happen. I'ce got a fix here if you're willing to test it.

Please clear your Adaptive_Mesh.cfg file and paste this in and try it again.

```yaml
# # # Klipper Adaptive Meshing # # #

# Heads up! If you have any other BED_MESH_CALIBRATE macros defined elsewhere in your config, you will need to comment out / remove them for this to work. (Klicky/Euclid Probe)
# You will also need to be sure that [exclude_object] is defined in printer.cfg, and your slicer is labeling objects.
# This macro will parse information from objects in your gcode to define a min and max mesh area to probe, creating an adaptive mesh!
# This macro will not increase probe_count values in your [bed_mesh] config. If you want richer meshes, be sure to increase probe_count. We recommend at least 5,5.

[gcode_macro BED_MESH_CALIBRATE]
rename_existing: _BED_MESH_CALIBRATE
gcode:

    {% set all_points = printer.exclude_object.objects | map(attribute='polygon') | sum(start=[]) %}                                # Gather all object points
    {% set bed_mesh_min = printer.configfile.settings.bed_mesh.mesh_min %}                                                          # Get bed mesh min from printer.cfg
    {% set bed_mesh_max = printer.configfile.settings.bed_mesh.mesh_max %}                                                          # Get bed mesh max from printer.cfg
    {% set probe_count = printer.configfile.settings.bed_mesh.probe_count %}                                                        # Get probe count from printer.cfg
    {% set verbose_enable = printer["gcode_macro _KAMP_Settings"].verbose_enable | abs %}                                           # Pull verbose setting from _KAMP_Settings
    {% set probe_dock_enable = printer["gcode_macro _KAMP_Settings"].probe_dock_enable | abs %}                                     # Pull probe dockable probe settings from _KAMP_Settings
    {% set attach_macro = printer["gcode_macro _KAMP_Settings"].attach_macro | string %}                                            # Pull attach probe command from _KAMP_Settings
    {% set detach_macro = printer["gcode_macro _KAMP_Settings"].detach_macro | string %}                                            # Pull detach probe command from _KAMP_Settings
    {% set mesh_margin = printer["gcode_macro _KAMP_Settings"].mesh_margin | float %}                                               # Pull mesh margin setting from _KAMP_Settings
    {% set fuzz_amount = printer["gcode_macro _KAMP_Settings"].fuzz_amount | float %}                                               # Pull fuzz amount setting from _KAMP_Settings
    {% set probe_count = probe_count if probe_count|length > 1 else probe_count * 2  %}                                             # If probe count is only a single number, convert it to 2. E.g. probe_count:7 = 7,7
    {% set max_probe_point_distance_x = ( bed_mesh_max[0] - bed_mesh_min[0] ) / (probe_count[0] - 1)  %}                            # Determine max probe point distance
    {% set max_probe_point_distance_y = ( bed_mesh_max[1] - bed_mesh_min[1] ) / (probe_count[1] - 1)  %}                            # Determine max probe point distance
    {% set x_min = all_points | map(attribute=0) | min | default(bed_mesh_min[0]) %}                                                # Set x_min from smallest object x point
    {% set y_min = all_points | map(attribute=1) | min | default(bed_mesh_min[1]) %}                                                # Set y_min from smallest object y point
    {% set x_max = all_points | map(attribute=0) | max | default(bed_mesh_max[0]) %}                                                # Set x_max from largest object x point
    {% set y_max = all_points | map(attribute=1) | max | default(bed_mesh_max[1]) %}                                                # Set y_max from largest object y point

    {% set fuzz_range = range((0) | int, (fuzz_amount * 100) | int + 1) %}                                                          # Set fuzz_range between 0 and fuzz_amount
    {% set adapted_x_min = (bed_mesh_min[0] + fuzz_amount - mesh_margin, x_min) | max - (fuzz_range | random / 100.0) %}            # Adapt x_min to margin and fuzz constraints
    {% set adapted_y_min = (bed_mesh_min[1] + fuzz_amount - mesh_margin, y_min) | max - (fuzz_range | random / 100.0) %}            # Adapt y_min to margin and fuzz constraints
    {% set adapted_x_max = (bed_mesh_max[0] - fuzz_amount + mesh_margin, x_max) | min + (fuzz_range | random / 100.0) %}            # Adapt x_max to margin and fuzz constraints
    {% set adapted_y_max = (bed_mesh_max[1] - fuzz_amount + mesh_margin, y_max) | min + (fuzz_range | random / 100.0) %}            # Adapt y_max to margin and fuzz constraints

    {% if adapted_x_min < x_min or adapted_y_min < y_min or adapted_x_max > x_max or adapted_y_max > y_max %}                       # If KAMP adjustments exceed config limits, remove adjustments and verbose message
        {% set adapted_x_min = x_min %}
        {% set adapted_y_min = y_min %}
        {% set adapted_x_max = x_max %}
        {% set adapted_y_max = y_max %}

        {% if verbose_enable == True %}
            {action_respond_info("KAMP adjustments would exceed defined bounds, running standard bed mesh.")}
        {% endif %}
    {% endif %}

[... cut ...]
    _BED_MESH_CALIBRATE mesh_min={adapted_x_min},{adapted_y_min} mesh_max={adapted_x_max},{adapted_y_max} ALGORITHM={algorithm} PROBE_COUNT={points_x},{points_y}

    {% if probe_dock_enable == True %}
        {detach_macro}                                                                                                              # Detach/stow a probe if the probe is stored somewhere outside of the print area
    {% endif %}                                                                                                                     # End of verbose
```

kyleisah, 2023-07-24:
This is just a stop-gap until I get a more elegant solution worked out. I think if the adjustments are going to exceed the configured bed mesh settings, KAMP should cancel the adjustments and just call a normal BED_MESH_CALIBRATE command. This is just to see if what I've got here will fix the issue in a pinch.

karlc28, 2023-07-27:
this was the issue i was facing i will test this in a few hours if it hasnt been done already

karlc28, 2023-07-31:
sorry only just tested today 
It still gives the same errors

karlc28, 2023-07-31:
[klippy (2).log](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12216161/klippy.2.log)

kyleisah, 2023-07-31:
@karlc28 Do you hapeen to have a photo of your printer? I'm trying to wrap my head around this still and I think that would help.

kyleisah, 2023-07-31:
I'll take another look at my fix and see if I can make that behave correctly, that definitely shouldn't be happening.

karlc28, 2023-08-15:
any luck

kyleisah, 2023-08-20:
This is now fixed in the main release, please update.

closed 2023-08-20 by kyleisah

VpopUnik, 2023-09-11:
I'm getting this error so I updated KAMP like you suggested (I had an older version which worked fine before). But it didn't solve the issue. First probe point and everything on the Left side of the bed is as -2,
If I do a BED_MESH I get this issue or if I print something that is on the left side of the bed. I can still print if I move the object closer to the middle/right.

BED_mesh from my printer.cfg:
[bed_mesh]
speed: 175
horizontal_move_z: 5
mesh_min: 29, 0
mesh_max: 222, 203
probe_count: 5,5
algorithm: bicubic
fade_start: 1
fade_end: 10
fade_target: 0

Move out of range: -2.000 45.000 5.000 [0.000]
Move out of range: -2.000 45.000 5.000 [0.000]
Move out of range: -2.000 45.000 5.000 [0.000]
24 | (173.0, 220.0) | (200.0, 200.0)
23 | (129.2, 220.0) | (156.2, 200.0)
22 | (85.5, 220.0) | (112.5, 200.0)
21 | (41.8, 220.0) | (68.8, 200.0)
**20 | (-2.0, 220.0) | (25.0, 200.0)
19 | (-2.0, 176.2) | (25.0, 156.2)**
18 | (41.8, 176.2) | (68.8, 156.2)
17 | (85.5, 176.2) | (112.5, 156.2)
16 | (129.2, 176.2) | (156.2, 156.2)
15 | (173.0, 176.2) | (200.0, 156.2)
14 | (173.0, 132.5) | (200.0, 112.5)
13 | (129.2, 132.5) | (156.2, 112.5)
12 | (85.5, 132.5) | (112.5, 112.5)
11 | (41.8, 132.5) | (68.8, 112.5)
**10 | (-2.0, 132.5) | (25.0, 112.5)
9 | (-2.0, 88.8) | (25.0, 68.8)**
8 | (41.8, 88.8) | (68.8, 68.8)
7 | (85.5, 88.8) | (112.5, 68.8)
6 | (129.2, 88.8) | (156.2, 68.8)
5 | (173.0, 88.8) | (200.0, 68.8)
4 | (173.0, 45.0) | (200.0, 25.0)
3 | (129.2, 45.0) | (156.2, 25.0)
2 | (85.5, 45.0) | (112.5, 25.0)
1 | (41.8, 45.0) | (68.8, 25.0)
**0 | (-2.0, 45.0) | (25.0, 25.0)**
bed_mesh: generated points
Index | Tool Adjusted | Probe
Generating new points...
KAMP adjustments successful. Happy KAMPing!
Adapted mesh bounds: (25.0, 25.0), (200.0, 200.0).
Fuzz amount is 0, mesh points not fuzzed.
Mesh margin is 0, margin not increased.
Default mesh bounds: (25.0, 25.0), (200.0, 200.0).
Adapted probe count: 5,5.
Default probe count: 5,5.
Algorithm: lagrange.
BED_MESH