kyleisah/Klipper-Adaptive-Meshing-Purging #137: Purge ignores brim

pOmelchenko, 2023-07-26:
**Describe the issue**
I was slice model with 20 mm of brim and purge macro ignore it. 

**What slicer are you using?**
Prusaslicer

**Troubleshooting**
disable gurge in config

**Screenshots**
[image]
[image]

**Klipper log file:**

**Sliced Gcode**
https://gist.github.com/pOmelchenko/8dc40d6c45dc1412e9258cba1aa3cf26

gullygossner, 2023-07-27:
I just ran into the same issue yesterday but I'm using orcaslicer.

kyleisah, 2023-07-31:
Hello! 

Just so you're aware, the purge is not ignoring the brim. The brim isn't labeled as an object by your slicer, so KAMP doesn't see it. I've been trying to convince @SoftFever to label these objects, but it can cause confusion when excluding an object from the frontend (mainsail/fluidd)

pOmelchenko, 2023-08-03:
But why? Klipper macro can't to check first layer for border assembly?

In my opinion for macro no need to know about any properties of g-code but only points of first layer. That information must be enough to build border and get coordinates to place purge

kyleisah, 2023-08-03:
...that's how KAMP works though. KAMP works off of objects labeled by your slicer to build the bounds of the print area. 

If the brim was labeled as an object, KAMP would see it, and correctly build the bounds of the print area.

kyleisah, 2023-08-03:
You are more than welcome to create a solution and make a PR for it to be added to KAMP, but it is my belief that anything a slicer is generating should be a labeled object.

pOmelchenko, 2023-08-03:
But sometimes we have G-code which isn't from slicer. Some calcs result, for example, pressure advance calculators in web. Where we don't have any labels.

I'm not sure about PR put I'll think about algorithm

closed 2023-08-05 by kyleisah