kyleisah/Klipper-Adaptive-Meshing-Purging #139: Can the position of adaptive clearing be adjusted?

skyaee, 2023-07-30:
Can the position of adaptive clearing be adjusted?
When printing small objects, clear the protruding height of the consumables, and it is easy to hit the nozzle. In the settings, I didn't find a place to adjust the clear position.

kyleisah, 2023-07-31:
I think I understand what you're saying, there is a proposed fix already for this in the other branch of this repo.

If you'd like to try it, change your line_purge to this:

```yaml

[gcode_macro LINE_PURGE]
description: A purge macro that adapts to be near your actual printed objects
gcode:
    # Get relevant printer params
    {% set travel_speed = (printer.toolhead.max_velocity) * 60 | float %}
    {% set cross_section = printer.configfile.settings.extruder.max_extrude_cross_section | float %}
    
    # Use firmware retraction if it is defined
    {% if printer.firmware_retraction is defined %}
        {% set RETRACT = G10 | string %}
        {% set UNRETRACT = G11 | string %}
    {% else %}
        {% set RETRACT = 'G1 E-.5 F2100' | string %}
        {% set UNRETRACT = 'G1 E.5 F2100' | string %}
    {% endif %}

    # Get purge settings from _Kamp_Settings
    {% set verbose_enable = printer["gcode_macro _KAMP_Settings"].verbose_enable | abs %}
    {% set purge_height = printer["gcode_macro _KAMP_Settings"].purge_height | float %}
    {% set tip_distance = printer["gcode_macro _KAMP_Settings"].tip_distance | float %}
    {% set purge_margin = printer["gcode_macro _KAMP_Settings"].purge_margin | float %}
    {% set purge_amount = printer["gcode_macro _KAMP_Settings"].purge_amount | float %}
    {% set flow_rate = printer["gcode_macro _KAMP_Settings"].flow_rate | float %}

    # Calculate purge origins and centers from objects
    {% set all_points = printer.exclude_object.objects | map(attribute='polygon') | sum(start=[]) %}    # Get all object points
    {% set purge_x_min = (all_points | map(attribute=0) | min | default(0)) %}                          # Object x min
    {% set purge_x_max = (all_points | map(attribute=0) | max | default(0)) %}                          # Object x max
    {% set purge_y_min = (all_points | map(attribute=1) | min | default(0)) %}                          # Object y min
    {% set purge_y_max = (all_points | map(attribute=1) | max | default(0)) %}                          # Object y max

    {% set purge_x_center = ([((purge_x_max + purge_x_min) / 2) - (purge_amount / 2), 0] | max) %}      # Create center point of purge line relative to print on X axis
    {% set purge_y_center = ([((purge_y_max + purge_y_min) / 2) - (purge_amount / 2), 0] | max) %}      # Create center point of purge line relative to print on Y axis

    {% set purge_x_origin = ([purge_x_min - purge_margin, 0] | max) %}                                  # Add margin to x min, compare to 0, and choose the larger
    {% set purge_y_origin = ([purge_y_min - purge_margin, 0] | max) %}                                  # Add margin to y min, compare to 0, and choose the larger

    # Calculate purge speed
    {% set purge_move_speed = (flow_rate / cross_section) * 60 | float %}

    {% if cross_section != 5 %}

        {action_respond_info("[Extruder] max_extrude_cross_section is not configured correctly, please set it to 5. Purge skipped.")}

    {% else %}

        {% if verbose_enable == True %}

        {action_respond_info("Moving filament tip {}mms".format(                                                                 
            (tip_distance),                                                                                      
        )) }
        {% endif %}

        {% if printer.firmware_retraction is defined %}
            {action_respond_info("KAMP purge is using firmware retraction.")}
        {% else %}
            {action_respond_info("KAMP purge is not using firmware retraction, it is recommended to configure it.")}
        {% endif %}

        {% if purge_y_origin > 0 %}
        
            {action_respond_info("KAMP purge starting at {}, {} and purging {}mm of filament, requested flow rate is {}mm/s3.".format(                                                                 
                (purge_x_center),
                (purge_y_origin),
[... 29 lines cut ...]
            G0 Z{purge_height * 2} F{travel_speed}                                              # Z hop

        {% else %}                                                                              # If there's room on X, purge along Y axis to the left of print area
            
            G92 E0                                                                              # Reset extruder
            G0 F{travel_speed}                                                                  # Set travel speed
            G90                                                                                 # Absolute positioning
            G0 X{purge_x_origin} Y{purge_y_center}                                              # Move to purge position
            G0 Z{purge_height}                                                                  # Move to purge Z height
            M83                                                                                 # Relative extrusion mode
            G1 E{tip_distance} F{purge_move_speed}                                              # Move filament tip
            G1 Y{purge_y_center + purge_amount} E{purge_amount} F{purge_move_speed}             # Purge line
            {RETRACT}                                                                           # Retract
            G92 E0                                                                              # Reset extruder distance
            M82                                                                                 # Absolute extrusion mode
            G0 Z{purge_height * 2} F{travel_speed}                                              # Z hop
            G0 Y{purge_y_center + purge_amount + 10} F{travel_speed}                            # Rapid move to break string

        {% endif %}
    
    {% endif %}

```

This adds a small rapid move at the end of the purge, which should break any stringing, and get the nozzle out of the way enough to not strike the purge line.

skyaee, 2023-07-31:
[image]

Thank you, I tried the code you shared, restarted printing, purge line, still very close to the print model, when printing the first layer, it is very easy to hit. Causes the first layer to fail to print.

[image]

kyleisah, 2023-07-31:
This is a very tough problem to fix because the Z height after the purge is your slicers first layer height setting, and also it depends on where your slicer is sending the printhead for the very first print move, and that's why the nozzle will sometimes have a chance to hit the purge line. 

Some things you can do to help this would be to have a single loop skirt around your print area 1mm from the print objects. (That's what I do) or you can try doing the voron logo purge instead because that purges at the corner of the print area versus in the front of it.

kyleisah, 2023-07-31:
Alternatively, I can make the rapid move at the end move a little bit in the Y direction to make sure the purge is out of the path of the nozzle. That might be the best thing to do.

kyleisah, 2023-07-31:
Please change the line_purge macro to this and try it again, this should fix it for sure. 

```yaml
[gcode_macro LINE_PURGE]
description: A purge macro that adapts to be near your actual printed objects
gcode:
    # Get relevant printer params
    {% set travel_speed = (printer.toolhead.max_velocity) * 60 | float %}
    {% set cross_section = printer.configfile.settings.extruder.max_extrude_cross_section | float %}
    
    # Use firmware retraction if it is defined
    {% if printer.firmware_retraction is defined %}
        {% set RETRACT = G10 | string %}
        {% set UNRETRACT = G11 | string %}
    {% else %}
        {% set RETRACT = 'G1 E-.5 F2100' | string %}
        {% set UNRETRACT = 'G1 E.5 F2100' | string %}
    {% endif %}

    # Get purge settings from _Kamp_Settings
    {% set verbose_enable = printer["gcode_macro _KAMP_Settings"].verbose_enable | abs %}
    {% set purge_height = printer["gcode_macro _KAMP_Settings"].purge_height | float %}
    {% set tip_distance = printer["gcode_macro _KAMP_Settings"].tip_distance | float %}
    {% set purge_margin = printer["gcode_macro _KAMP_Settings"].purge_margin | float %}
    {% set purge_amount = printer["gcode_macro _KAMP_Settings"].purge_amount | float %}
    {% set flow_rate = printer["gcode_macro _KAMP_Settings"].flow_rate | float %}

    # Calculate purge origins and centers from objects
    {% set all_points = printer.exclude_object.objects | map(attribute='polygon') | sum(start=[]) %}    # Get all object points
    {% set purge_x_min = (all_points | map(attribute=0) | min | default(0)) %}                          # Object x min
    {% set purge_x_max = (all_points | map(attribute=0) | max | default(0)) %}                          # Object x max
    {% set purge_y_min = (all_points | map(attribute=1) | min | default(0)) %}                          # Object y min
    {% set purge_y_max = (all_points | map(attribute=1) | max | default(0)) %}                          # Object y max

    {% set purge_x_center = ([((purge_x_max + purge_x_min) / 2) - (purge_amount / 2), 0] | max) %}      # Create center point of purge line relative to print on X axis
    {% set purge_y_center = ([((purge_y_max + purge_y_min) / 2) - (purge_amount / 2), 0] | max) %}      # Create center point of purge line relative to print on Y axis

    {% set purge_x_origin = ([purge_x_min - purge_margin, 0] | max) %}                                  # Add margin to x min, compare to 0, and choose the larger
    {% set purge_y_origin = ([purge_y_min - purge_margin, 0] | max) %}                                  # Add margin to y min, compare to 0, and choose the larger

    # Calculate purge speed
    {% set purge_move_speed = (flow_rate / 5.0) * 60 | float %}

    {% if cross_section < 5 %}

        {action_respond_info("[Extruder] max_extrude_cross_section is insufficient for purge, please set it to 5 or greater. Purge skipped.")}

    {% else %}

        {% if verbose_enable == True %}

        {action_respond_info("Moving filament tip {}mms".format(                                                                 
            (tip_distance),                                                                                      
        )) }
        {% endif %}

        {% if printer.firmware_retraction is defined %}
            {action_respond_info("KAMP purge is using firmware retraction.")}
        {% else %}
            {action_respond_info("KAMP purge is not using firmware retraction, it is recommended to configure it.")}
        {% endif %}

        {% if purge_y_origin > 0 %}
        
            {action_respond_info("KAMP purge starting at {}, {} and purging {}mm of filament, requested flow rate is {}mm/s3.".format(                                                                 
                (purge_x_center),
                (purge_y_origin),
                (purge_amount),
                (flow_rate),
            )) }
[... 30 lines cut ...]
        {% else %}                                                                              # If there's room on X, purge along Y axis to the left of print area
            
            G92 E0                                                                              # Reset extruder
            G0 F{travel_speed}                                                                  # Set travel speed
            G90                                                                                 # Absolute positioning
            G0 X{purge_x_origin} Y{purge_y_center}                                              # Move to purge position
            G0 Z{purge_height}                                                                  # Move to purge Z height
            M83                                                                                 # Relative extrusion mode
            G1 E{tip_distance} F{purge_move_speed}                                              # Move filament tip
            G1 Y{purge_y_center + purge_amount} E{purge_amount} F{purge_move_speed}             # Purge line
            {RETRACT}                                                                           # Retract
            G0 Y{purge_y_center + purge_amount + 10} Y{purge_y_origin + 5} F{travel_speed}      # Rapid move to break string
            G92 E0                                                                              # Reset extruder distance
            M82                                                                                 # Absolute extrusion mode
            G0 Z{purge_height * 2} F{travel_speed}                                              # Z hop

        {% endif %}

        RESTORE_GCODE_STATE NAME=Prepurge_State                                                 # Restore gcode state
    
    {% endif %}

```

I added a 5mm movement in Y to the rapid move to clear the purge line and force the nozzle to be out of the way

skyaee, 2023-08-03:
After updating the new version yesterday, I found that both the line and the voron logo will overlap with the loop skirt around your print area. As a result, both modes cannot be used normally.

kyleisah, 2023-08-03:
How many loops are you using?

The slicer does not label these objects so KAMP can't adjust for them which is frustrating.

skyaee, 2023-08-03:
3 loops   10mm

dougshell, 2023-08-19:
why are you doing a skirt anyway? isn't it to normalize pressure in the nozzle prior to printing which kamp already does

kyleisah, 2023-08-20:
This is a slicer-side issue. Please ask the creator of your slicer to label skirts as objects.

closed 2023-08-20 by kyleisah