kyleisah/Klipper-Adaptive-Meshing-Purging #140: Firmware Retraction > not an issue just some questions

Serox13, 2023-07-31:
Hey,
  I have been using adaptive purge and mesh for a couple of weeks i have tried it with and without firmware retraction enabled. Besides having extra features in the slicer retraction settings available when its NOT enabled, i haven't noticed any issues, why is it recommended that firmware retraction is used with KAMP? i have looked and read all over your documentation and it just says it is recommended to have it on. 

Are there any advantages to using firmware retraction with KAMP?
Sorry if these seem like basic questions thanks for your time.

kyleisah, 2023-07-31:
Hey! No worries, so the usage of firmware retraction has a lot of benefits:
- smaller gcode files (`G10` and `G11` replace `G1 E-.4 F8000` which helps make gcode files a bit lighter)
- live-tuning your retractions rather than being stuck with what you sliced (helps if some filament got a little wet, you can bump your settings during a print to retract a little more)
- offloading retract settings on a "per-machine" basis, so the exact same gcode can be ran on many different machines and they will all use the correct defined retraction.

This *does* hinder some slicer features though, like wipe, coast, etc, which is no fun.

The great thing is, having the [firmware_retraction] config block **ONLY** affects `G10` and `G11` commands. So you *can* have both firmware retraction, *and* slicer-based retraction enabled and not even see a difference.

I mostly have it in there because a lot of folks use bowden extruder setups which require a lot more retraction length than a direct-drive extruder.

The default KAMP purge configs use a DD extruder retraction by default if there's no firmware retraction enabled, but if your purges are super stringy or ugly or ooze a lot because you use a bowden and the retractions aren't right, you can use firmware retraction to make KAMP use the retraction you need for your specific machine.

That's quite a lot of information lol, I hoped that answered your question!

closed 2023-08-05 by kyleisah

larkinnn, 2023-08-08:
how do i find out the values i need just play around?