kyleisah/Klipper-Adaptive-Meshing-Purging #141: KAMP settings can't reproduce previous (before update) results

uki-3D, 2023-07-31:
Hello. Today I decided to reinstall KAMP; previously I had both configuration files added manually as it used to be in the tutorial.
Now I can't achieve the results I had before. If I set variable_flow_rate: 6 then I get much too much purge. If I set variable_flow_rate: 2 or 1, the result is acceptable but it takes an awful long time to purge...
Here are my previous settings:

```
variable_adaptive_enable: True      # Change to False if you'd like the purge to be in the same spot every print
variable_z_height: 0.4              # Height above the bed to purge
variable_purge_amount: 15           # Amount of filament in millimeters to purge
variable_line_length: 40            # Overall desired length of purge line in millimeters, around 1/5th of X axis length is a good starting value
variable_flow_rate: 6               # Desired flow rate in mm3/s (Around 12 for standard flow hotends, around 24 for high flow hotends)
variable_x_default: 10              # Default X location to purge. If adaptive_enable is True, this is overwritten
variable_y_default: 10              # Default Y location to purge. If adaptive_enable is True, this is overwritten
variable_distance_to_object_y: 10   # Y distance in millimeters away from the print area for purging. Must be less than or equal to y_default if adaptive_enable is False
```

Why can't I achieve the previous result with the new settings?
I noticed that one parameter is missing: variable_line_length: xx.
Was it intentionally removed?
Thank you very much for this project, as for me EVERY printer should have it!

kyleisah, 2023-08-01:
I think what might be going on is you're using the old macros still rather than the new intended method. 

All settings and [include]s for KAMP are stored inside of KAMP_Settings.cfg. 

You'll need to remove any old macros / KAMP configs from before and set up KAMP again. The old macros each had settings inside the macros and I wanted to keep all relevant params in one place to make it easier to provide support. 

Hope that helps!

kyleisah, 2023-08-01:
Also, line length has been removed and is calculated from the extrude amount requested. This helps make the purge line the same overall shape across all machines. 

You can dig around in the macro to find out more, but it calculates flow from your extruder settings and nozzle size, etc. It seems to be working pretty well for most.

uki-3D, 2023-08-01:
Thanks for the answer. 
I deleted the old macros before installing the new version.
Unfortunately, I still can't get anything remotely similar to what I had before. However I set "purge_amount" it always spits out the same amount of filament per millimeter which gives a big thick line where the nozzle goes at the very end and pulls the strings behind it. If I set the "flow_rate" to a lower value, the line is acceptable, but the whole process takes about 3 minutes.
Well, I guess I have to live with it.

Top - before.
Bottom - after.
[image]

closed 2023-08-05 by kyleisah