kyleisah/Klipper-Adaptive-Meshing-Purging #144: Adaptive bed mesh not working for Prusaslicer 2.6

jordandrako, 2023-08-01:
**Describe the issue**
When sending the same models in both Superslicer and Prusaslicer, the adaptive bed mesh macro doesn't detect the model bounds on the Prusaslicer gcode. This is because Prusaslicer is outputing the exclude_object definitions AFTER the custom start print gcode.

**What slicer are you using?**
Prusaslicer 2.6 and Superslicer 2.5.59

**Troubleshooting**
I moved the exclude object definitions to right before my START_PRINT macro and it worked as expected.

**Sliced Gcode**
Here's the 3 files I tested: the working superslicer gcode, broken prusaslicer gcode, and fixed prusaslicer gcode.
https://gist.github.com/jordandrako/bec5aba0cbf80cf1bef140a9e7d71edc

**Additional context**
Since the adaptive mesh isn't getting the bounds of the objects, the line purge and smart park are also not working, obviously.

werkstrom, 2023-08-03:
@jordandrako Did you manage to do that with a Post Processing script set up in Prusa Slicer (Print Settings -> Output Options -> Post Processing Scripts) or did you do it manually by editing the file?

kyleisah, 2023-08-03:
This is a known issue with the [exclude_object] processor in moonraker. You can add `M117` before Print_Start in your slicers start gcode section to force the processor to add the lines before print_start. 

The processor injects the definition macros after the first gcode command it sees, and on some machines, for reasons beyond me, it just doesn't lol. 

During dev we found using the "clear display" command (M117) was a nice non intrusive command to make it happen.

werkstrom, 2023-08-03:
Yep @kyleisah , just found that out and wanted to share... You beat me to it 😉

jordandrako, 2023-08-03:
Thanks @kyleisah! So the actual reason is that Superslicer output a single line of gcode before my custom start gcode, and it's moonraker that processed that and inserted the exclude object definitions, where Prusaslicer output my custom start gcode before anything else. Got it.

Well I'll close this issue since it's technically an upstream bug, and your workaround is perfectly unobtrusive.

closed 2023-08-03 by jordandrako

Bomberski, 2023-08-08:
M117 worked... first time using this and wonder why it always did a full mesh. No matter what size the object was... thanks it actually works great now