kyleisah/Klipper-Adaptive-Meshing-Purging #145: Adaptive Mesh for Delta Printers

jpbnto, 2023-08-02:
**Describe the issue*
I think Delta printers is not yet supported

**What slicer are you using?**
Orcaslicer

**Troubleshooting**
I tried adjusting max mesh radius, same result

**Screenshots**
[image]

**Klipper log file:**
[klippy.log](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12235181/klippy.log)

**Sliced Gcode**
[Cylinder left.gcode.txt](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12235210/Cylinder.left.gcode.txt)

**Additional context**
Hope Delta Printes will be supoorted

kyleisah, 2023-08-02:
KAMP has no delta support. I had something that theoretically worked on delta printers, but I don't have a delta so I can't troubleshoot or test anything.

kyleisah, 2023-08-03:
@jpbnto are you on discord by any chance? I have something you can try.

jpbnto, 2023-08-03:
Yes i am. 

Look me up jpbnto

closed 2023-08-05 by kyleisah

pradeep1080, 2023-08-14:
Hey @kyleisah I also want to try adaptive mesh in delta printers. Can I get any help?