kyleisah/Klipper-Adaptive-Meshing-Purging #147: Adaptive bed issue

S1mone01, 2023-08-03:
when I try to print a large object it happens to cancel it while it carries out the first layer but every time to restart the print it is necessary to redo all the adaptive bed mesh which takes a long time. Is there a way to momentarily save the bed mesh of that gcode and when it restarts it doesn't have to do everything again?

kyleisah, 2023-08-05:
There is an issue template. Please upload a klipper log and give me a description of the error you saw so I can correctly troubleshoot.

kyleisah, 2023-08-05:
As far as saving and loading meshes, read the klipper documentation. 

https://www.klipper3d.org/Bed_Mesh.html#profiles

greendog99, 2023-08-05:
I think the original poster is asking for a new feature, not reporting an issue. The feature request is one I have as well:

If re-printing a file with the exact same dimensions as the previous file, do not re-create the mesh from scratch -- just use the existing in-memory mesh.

greendog99, 2023-08-11:
> > I think the original poster is asking for a new feature, not reporting an issue. The feature request is one I have as well:
> > If re-printing a file with the exact same dimensions as the previous file, do not re-create the mesh from scratch -- just use the existing in-memory mesh.
> 
> Maybe just remove bedmesh calibration gcode?

Definitely an option, if you don't mind editing your gcode when restarting a failed print or printing a second copy of something.

I think I'll try forking and adding an option which compares the computed x_min / x_max / y_min / y_max to the previously meshed ones (saved to some new variables) and skips the bed mesh if they match. That way if I need to reprint something for any reason, it won't re-mesh when I restart each print.

S1mone01, 2023-08-11:
> Definitely an option, if you don't mind editing your gcode when restarting a failed print or printing a second copy of something.
>
> I think I'll try forking and adding an option which compares the [...]

Yes, that's what interests me. If you managed to accomplish this option, could you please share it?

closed 2023-08-11 by kyleisah

greendog99, 2023-08-23:
> Yes, that's what interests me. If you managed to accomplish this option, could you please share it?

https://gist.github.com/greendog99/0d5f1da3dcbebf862412502b04100bcf

I just put this new macro together and it seems to do the trick. In my START_GCODE I now call KAMP_CONDITIONALLY instead of calling BED_MESH_CALIBRATE. The new macro remembers the dimensions of the print, so if you run additional prints of the same size, it will know that it can skip the KAMP bed mesh. I've only tested it once, so there may be issues I'm unaware of.