kyleisah/Klipper-Adaptive-Meshing-Purging #148: Update for deprecated relative_reference_index parameter

davisgoodman, 2023-08-04:
As the new klipper version is deprecating the relative_referene_index parameter, everytime I run an adaptive bed mesh I get the notification of the deprecated parameter..

Still works but I believe the current macro doesn't take care of the new zero_reference_position parameter

Is their an update that will take care of using the new zero_reference_position instead of the deprecated parameter or is the current code able to adapt without any modifications.

I've updated in the printer.cfg the [bed_mesh] section with the new parameter but I suppose this would need to be set for each bed mesh

I'm using SuperSlicer

AudreyAP, 2023-08-04:
Remove relative_reference_index and add zero_reference_position to your printer.cfg.

KAMP has nothing to do with it, and nothing needs to be done per bed mesh. That's the point of the new setting :)

davisgoodman, 2023-08-04:
Great, 

 

Thanks for the reply

closed 2023-08-05 by kyleisah