kyleisah/Klipper-Adaptive-Meshing-Purging #149: After adding adaptive mesh and purge it alwasys does both twice

Wes910, 2023-08-06:
**Describe the issue**
I added adaptive mesh and purge, When I start a print it wants to do them twice before the prints starts

**What slicer are you using?**
Superslicer

**Troubleshooting**
looked at printer cnfg and slicer settings

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Klipper log file:**
Please add klipper log file here...
[klippy.log](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12269761/klippy.log)

**Sliced Gcode**
Please add sliced gcode file you are using here...
[Calibration cube.txt](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12269741/Calibration.cube.txt)

**Additional context**
Thanks for the help

kyleisah, 2023-08-06:
[image]

Looks like for some reason, PRINT_START is being called twice by your slicer. Once as starting gcode, and again by filament gcode.

Wes910, 2023-08-07:
Im not sure why superslicer is doing this?

[image]

Wes910, 2023-08-07:
I got it sorted, evidently super slicer has a custom g code sections for the filament and printer. I had the code in both, I must not realised I was in one and not the other

closed 2023-08-07 by Wes910