kyleisah/Klipper-Adaptive-Meshing-Purging #150: Not Working

Immrtlchaos, 2023-08-06:
**Describe the issue**
Adaptive Meshing isn't working at all. When I first installed it on a fresh klipper install, it worked shortly after. However after working out some issues in my printer config file and getting everything to work like it was suppsed to, it doesn't work at all. Do I need to add something to my G-Code in Prusaslicer? I have even reinstalled KAMP to see if it could straighten it out but have had no luck. This is my Print start G-Code:

M104 S0 ; Stops PS/SS from sending temp waits separately
M140 S0
PRINT_START BED=[first_layer_bed_temperature] HOTEND=[first_layer_temperature[initial_extruder]]
PURGE_LINE

kyleisah, 2023-08-06:
Could you please upload a klipper log?

Immrtlchaos, 2023-08-06:
[logs-20230800-144049.zip](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12270280/logs-20230800-144049.zip)

Immrtlchaos, 2023-08-06:
For some reason I can only download a zip file of my Klippy log

kyleisah, 2023-08-06:
I think that's because your klipper log is very large. Happens sometimes. I'll take a look at this next chance I get

kyleisah, 2023-08-20:
Can you explain what you mean by "It doesn't work at all?" Just so I can narrow down my search

Immrtlchaos, 2023-08-20:
Crap. I forgot to respond. Found the issue. Had a line commented out in my Print Start macros. Sorry about that.

kyleisah, 2023-08-20:
Lol no worries dude. I've been busy too, sorry to keep you waiting but I'm glad it's all good now.

closed 2023-08-20 by kyleisah