kyleisah/Klipper-Adaptive-Meshing-Purging #151: Surface mode in Cura

maziarzm, 2023-08-08:
The adaptive mesh doesn't work when I use the "Surface mode: Surface" function in Cura.
Normally, I don't use this function and everything works fine. I downloaded ready-made profiles from an RC aircraft creator. 
I'm printing parts with LW-PLA using those profiles. This option is enabled in these profiles. When I start the print, the printer tries to move beyond the bed's limits and gives an error.

kyleisah, 2023-08-12:
Where's the gcode file and klipper log?

maziarzm, 2023-08-13:
I apologize, I submitted the report while on vacation. I have attached the files.

kyleisah, 2023-08-16:
It looks like this premade profile you're using doesn't label objects, or at least moonraker isn't detecting that objects are needing to be processed.

kyleisah, 2023-08-16:
```yaml

# This macro will allow stop-gap functionality of KAMP with Surface mode in Cura. Please add the following to Cura's 
# Start-Gcode section *before* PRINT_START is called, like so:
# .
# .
# MESH_CONFIG X0=[MINX] Y0=[MINY] X1=[MAXX] Y1=[MAXY]
# .
# PRINT_START EXTRUDER_TEMP=[first_layer_temperature] BED_TEMP=[first_layer_bed_temperature]
# .

# Then add this macro to your printer configs. 

[gcode_macro MESH_CONFIG]
description: Stop-gap macro for converting Cura's surface mode print area into an object for exclude_object
gcode:
    {% set X0 = params.X0 | float %}                # Print Area X Min
    {% set X1 = params.X1 | float %}                # Print Area X Max
    {% set Y0 = params.Y0 | float %}                # Print Area Y Min
    {% set Y1 = params.Y1 | float %}                # Print Area Y Max
    {% set center_x = ((X0 + X1) / 2) | float %}    # Print Area X Center
    {% set center_y = ((Y0 + Y1) / 2) | float %}    # Print Area Y Center

    EXCLUDE_OBJECT_DEFINE NAME=s3d_build_volume CENTER={center_x},{center_y} POLYGON=[[{X0},{Y0}],[{X0},{Y1}],[{X1},{Y1}],[{X1},{Y0}]]
```

kyleisah, 2023-08-16:
You can try the above patch, but I can't promise it will even work, as I don't use Cura.

kyleisah, 2023-08-20:
No updates, no fix I can provide due to slicer-side issue

closed 2023-08-20 by kyleisah