kyleisah/Klipper-Adaptive-Meshing-Purging #152: Error // [Extruder] max_extrude_cross_section is insufficient for purge, please set it to 5 or greater. Purge skipped.

weruhgfderfg, 2023-08-09:
I use cura 5.4 with klipper creality sonic pad (latest update with exclude objects) when ever i start a print the purge line just doesnt print and skips

truelecter, 2023-08-09:
>**Note:**
It is required to add `max_extrude_cross_section: 5` to your `[extruder]` config to allow effective purging to be possible. KAMP will warn you if you forget to set this value, and skip the purge so the printer will not be halted. It is also recommended to set up [firmware_retraction](https://www.klipper3d.org/Config_Reference.html?h=retract#firmware_retraction) inside of klipper so KAMP can use the correct retration settings for your machine. If this is not found, KAMP will warn you, and use reasonable direct-drive extruder values to complete the purge.

I guess you forgot to add this setting to the extruder configuration

weruhgfderfg, 2023-08-10:
@truelecter it works now tysm!

closed 2023-08-11 by kyleisah