kyleisah/Klipper-Adaptive-Meshing-Purging #153: Orca slicer error

ManuelPeiro, 2023-08-12:
**Describe the issue**
When I send a sliced file to print with ORCA, I get this error on the console

`!! Error evaluating 'gcode_macro BED_MESH_CALIBRATE:gcode': UndefinedError: 'dict object' has no attribute 'polygon'

!! Error evaluating 'gcode_macro BED_MESH_CALIBRATE:gcode': UndefinedError: 'dict object' has no attribute 'polygon'`

**What slicer are you using?**
Orcaslicer latest version

**Troubleshooting**
Error message and print dont start.
With Prusa Slic3r, works fine

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Klipper log file:**
Please add klipper log file here...
[klippy KAMP ERROR.log](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12326661/klippy.KAMP.ERROR.log)

**Sliced Gcode**
Ple
[ORCA_Petg- Bronco taller 02 Gears.zip](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12326667/ORCA_Petg-.Bronco.taller.02.Gears.zip)
ase add sliced gcode file you are using here...

**Additional context**
Add any other context about the problem here.

kyleisah, 2023-08-12:
Looks like you're using an old version of KAMP. Can you remove all of that and go through the install again and try?

ManuelPeiro, 2023-08-16:
Yes...

I dont know why... But its working perfect now...

Thx

closed 2023-08-20 by kyleisah