kyleisah/Klipper-Adaptive-Meshing-Purging #155: Error evaluating 'gcode_macro BED_MESH_CALIBRATE:gcode': TypeError: bad operand type for abs(): 'Undefined'

deonkretch, 2023-08-13:
**Describe the issue**
File sliced with Orca Slicer - I get this error:

**7:24 PM
Error evaluating 'gcode_macro BED_MESH_CALIBRATE:gcode': TypeError: bad operand type for abs(): 'Undefined'**

**What slicer are you using?**
Orca Slicer 1.6.4

**Troubleshooting**
Restarted printer multiple times.

**Klipper log file:**
[klippy (3).log](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12330032/klippy.3.log)

**Sliced Gcode**
[benchy.gcode.zip](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12330036/benchy.gcode.zip)

**Additional context**
No additional context.

kyleisah, 2023-08-16:
Did you delete the `variable_probe_dock_enable` line rather than set it to false?

kyleisah, 2023-08-20:
no replies

closed 2023-08-20 by kyleisah

kyleisah, 2023-08-21:
@mairisskuja i think that having it the way it is now makes sense. Nowhere in the documentation does it tell you to comment it out or delete it.