kyleisah/Klipper-Adaptive-Meshing-Purging #156: Move nozzle out after purge

mrodyhin, 2023-08-15:
In the current LINE_PURGE macro the last command is doing Z hop, and releases control to the next gcode.
The issue is that it's very likely, that the first move command generated by slicer, is move Z to the first layer height.
That results in a wired behaviour, that nozzle goes up, than back down, and than starts moving by XY axis thought the purge line.

Here is what i got: 
[image]

I tried to fix it by adding additional move in my print start macro after the purge, but i doesn't always work, since the purge macro can place the line on X or Y axis.

The proposal is to add additional XY move towards the modal, to move nozzle out of the purge line.

kyleisah, 2023-08-16:
Hey there! Just letting you know this has already been fixed and will eventually be pushed in the next update.

mrodyhin, 2023-08-17:
That's a great news, thank you very much.

kyleisah, 2023-08-20:
This is now in the main release. :)

closed 2023-08-20 by kyleisah